MarlinFirmware/Marlin #16001: [BUG] SKR 1.3 Conflict between case light and Fan1 pins

mvoss96, 2019-11-25:
In the pin definition file for the skr1.3 board (pins_BIGTREE_SKR_V1.3.h) the following code automatically assigns FAN1 to P2_04 (Heater 1) if only one extruder is active:
`#if HOTENDS == 1
  #ifndef FAN1_PIN
    #define FAN1_PIN       P2_04
  #endif`

This creates a conflict if you want to use the H1 output to drive case lights.

If you enable #define CASE_LIGHT_ENABLE and set #define CASE_LIGHT_PIN P2_04 it compiles fine but you get weird behavior where the output of H1 is controlled from case light as well as Fan1 settings rendering both settings unusable.

A possible fix would be to change the code above so that P2_04 only gets assigned to fan1 if no other option uses the pin already. F.e:

`#if HOTENDS == 1 && CASE_LIGHT_PIN != P2_04
  #ifndef FAN1_PIN
    #define FAN1_PIN       P2_04
  #endif`

shitcreek, 2019-11-26:
I wouldn't say this is a bug. Pins are something users should tweak to fit their specific needs. For instance `P2_04` could be used to trigger a laser/spindle. I'd just set the `FAN1_PIN` to -1

boelle, 2019-11-26:
clsoing since this is not a bug as such and tehre is no config supplied

if its a bug please follow the template and make sure to include what is required

closed 2019-11-26 by boelle