MarlinFirmware/Marlin #16002: MMU2 fails to build on Marlin 11/25/2019

tedukes, 2019-11-26:
### Description

Compiling .pio\build\LPC1768\src\src\gcode\feature\prusa_MMU2\M403.cpp.o
Marlin\src\gcode\feature\pause\M701_M702.cpp: In static member function 'static void GcodeSuite::M701()':
Marlin\src\gcode\feature\pause\M701_M702.cpp:104:10: error: 'class MMU2' has no member named 'load_filament_to_nozzle'; did you mean 'set_filament_type'?
     mmu2.load_filament_to_nozzle(target_extruder);
          ^~~~~~~~~~~~~~~~~~~~~~~
          set_filament_type
Marlin\src\gcode\feature\pause\M701_M702.cpp: In static member function 'static void GcodeSuite::M702()':
Marlin\src\gcode\feature\pause\M701_M702.cpp:208:10: error: 'class MMU2' has no member named 'unload'
     mmu2.unload();
          ^~~~~~
*** [.pio\build\LPC1768\src\src\gcode\feature\pause\M701_M702.cpp.o] Error 1

### Steps to Reproduce
**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files. 

Download and build current version of Marlin w/MMU2 support
**Expected behavior:** [What you expect to happen]

Compiling .pio\build\LPC1768\src\src\gcode\feature\prusa_MMU2\M403.cpp.o
Marlin\src\gcode\feature\pause\M701_M702.cpp: In static member function 'static void GcodeSuite::M701()':
Marlin\src\gcode\feature\pause\M701_M702.cpp:104:10: error: 'class MMU2' has no member named 'load_filament_to_nozzle'; did you mean 'set_filament_type'?
     mmu2.load_filament_to_nozzle(target_extruder);
          ^~~~~~~~~~~~~~~~~~~~~~~
          set_filament_type
Marlin\src\gcode\feature\pause\M701_M702.cpp: In static member function 'static void GcodeSuite::M702()':
Marlin\src\gcode\feature\pause\M701_M702.cpp:208:10: error: 'class MMU2' has no member named 'unload'
     mmu2.unload();
          ^~~~~~
*** [.pio\build\LPC1768\src\src\gcode\feature\pause\M701_M702.cpp.o] Error 1
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3889546/Marlin.zip)

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

RudyBenner, 2019-11-26:
Line 2765 in configuration_adv.h

ellensp, 2019-11-26:
Agreed #define MMU2_MENUS is disabled in Configuration_adv.h

tedukes, 2019-11-26:
^%$%&*!!

Thanks!! 

Missed one. I always use new config files rather that just copying. It now builds but lots of 'warnings'.

Marlin\src\feature\prusa_MMU2\mmu2.cpp: In static member function 'static void MMU2::mmu_loop()':
Marlin\src\feature\prusa_MMU2\mmu2.cpp:177:27: warning: format '%u' expects argument of type 'unsigned int*', but argument 3 has type 'int16_t*' {aka 
'short int*'} [-Wformat=]
         sscanf(rx_buffer, "%uok\n", &version);
                           ^~~~~~~~  ~~~~~~~~
Marlin\src\feature\prusa_MMU2\mmu2.cpp:188:27: warning: format '%u' expects argument of type 'unsigned int*', but argument 3 has type 'int16_t*' {aka 
'short int*'} [-Wformat=]
         sscanf(rx_buffer, "%uok\n", &buildnr);
                           ^~~~~~~~  ~~~~~~~~
Marlin\src\feature\prusa_MMU2\mmu2.cpp: In static member function 'static void MMU2::manage_response(bool, bool)':
Marlin\src\feature\prusa_MMU2\mmu2.cpp:593:27: warning: 'resume_hotend_temp' may be used uninitialized in this function [-Wmaybe-uninitialized]
       if (turn_off_nozzle && resume_hotend_temp) {

-----------------------------------------------------------------------------------------------------

Marlin\src\lcd\menu\menu_temperature.cpp: In function 'void menu_temperature()':
Marlin\src\lcd\menu\menu_temperature.cpp:215:50: warning: array subscript 1 is above array bounds of 'uint8_t [1]' {aka 'unsigned char [1]'} [-Warray-bounds]
       editable.uint8 = thermalManager.fan_speed[1];
                        ~~~~~~~~~~~~~~~~~~~~~~~~~~^
Marlin\src\lcd\menu\menu_temperature.cpp:225:50: warning: array subscript 2 is above array bounds of 'uint8_t [1]' {aka 'unsigned char [1]'} [-Warray-bounds]
       editable.uint8 = thermalManager.fan_speed[2];
                        ~~~~~~~~~~~~~~~~~~~~~~~~~~^
Marlin\src\lcd\menu\menu_tune.cpp: In function 'void menu_tune()':
Marlin\src\lcd\menu\menu_tune.cpp:164:50: warning: array subscript 1 is above array bounds of 'uint8_t [1]' {aka 'unsigned char [1]'} [-Warray-bounds]
       editable.uint8 = thermalManager.fan_speed[1];
                        ~~~~~~~~~~~~~~~~~~~~~~~~~~^
Marlin\src\lcd\menu\menu_tune.cpp:174:50: warning: array subscript 2 is above array bounds of 'uint8_t [1]' {aka 'unsigned char [1]'} [-Warray-bounds]
       editable.uint8 = thermalManager.fan_speed[2];

------------------------------------------------------------------------------------------------------

.pio\libdeps\LPC1768\U8glib-HAL_ID1932\src\clib\u8g_com_i2c.c:44:16: warning: 'u8g_i2c_opt' defined but not used [-Wunused-variable]
 static uint8_t u8g_i2c_opt;   /* U8G_I2C_OPT_NO_ACK, SAM: U8G_I2C_OPT_DEV_1 */

RudyBenner, 2019-11-26:
Can you close?

tedukes, 2019-11-26:
What about the warnings? Does that mean something is not going work? Can't test the build right now as I'm in the middle of an 18 hour lithophane. I can test this evening when I get home. The print should be finished by then.

RudyBenner, 2019-11-26:
I see some new commits related to MM2.

boelle, 2019-11-26:
warnings do not mean anything, what matters is if the compile is a success

we can reopen if it does not work even thou build is a success

closed 2019-11-26 by boelle