MarlinFirmware/Marlin #16007: [BUG] [Bugfix 2.0.x] Delta printer Calibrate X/Y/Z strange behaviour

vijexa, 2019-11-26:
### Description

When pressing Calibrate X(/Y/Z) in Delta Calibration printer starts to shake printhead in direction of corresponding tower and reboots after few seconds.

### Steps to Reproduce
Configuration files: [Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3891578/Configuration.zip)
Hardware: Anycubic Kossel Linear Plus with Bigtreetech SKR v1.3 board

1. Auto Home
2. Go to Configuration > Delta Calibration
3. Press Calibrate X(/Y/Z)

**Expected behavior:** Printer starts to calibrate corresponding axis (not sure though, using this features for the first time and can't find any info)

**Actual behavior:** Printer starts to violently shake printhead in direction of corresponding tower and reboots after few seconds

#### Additional Information

Marlin is downloaded today (26th of November)  from bugfix-2.0.x branch.
Here is a video I recorded: https://www.youtube.com/watch?v=qgJ1AZSIq4c (there is a probe attached, without a probe behaviour is the same)

sjasonsmith, 2019-11-26:
I was able to reproduce this even with code prior to my recent delta probing changes (b904ba0f291cdcb9759464f064f85ba066b778ce).

The center 

Something very strange is happening with this. I added some serial prints to the code, and I see the prints from the command attempting to execute multiple times, even though it was only selected once. I then see "cold extrusion prevented" before the controller resets.

```
_goto_tower_x
goto_tower @ 210.00
tower_vec (coefficients) = { -0.87, -0.50, }
_man_probe_pt = { -87.47, -50.50, }
do_blocking_move_to_xy_z = { -87.47, -50.50, 5.00} fr_mm_s=0.00
_goto_tower_x
goto_tower @ 210.00
tower_vec (coefficients) = { -0.87, -0.50, }
_man_probe_pt = { -87.47, -50.50, }
do_blocking_move_to_xy_z = { -87.47, -50.50, 5.00} fr_mm_s=0.00
_goto_tower_x
goto_tower @ 210.00
tower_vec (coefficients) = { -0.87, -0.50, }
_man_probe_pt = { -87.47, -50.50, }
do_blocking_move_to_xy_z = { -87.47, -50.50, 5.00} fr_mm_s=0.00
_goto_tower_x
goto_tower @ 210.00
tower_vec (coefficients) = { -0.87, -0.50, }
_man_probe_pt = { -87.47, -50.50, }
do_blocking_move_to_xy_z = { -87.47, -50.50, 5.00} fr_mm_s=0.00
_goto_tower_x
goto_tower @ 210.00
tower_vec (coefficients) = { -0.87, -0.50, }
_man_probe_pt = { -87.47, -50.50, }
do_blocking_move_to_xy_z = { -87.47, -50.50, 5.00} fr_mm_s=0.00
_goto_tower_x
goto_tower @ 210.00
tower_vec (coefficients) = { -0.87, -0.50, }
_man_probe_pt = { -87.47, -50.50, }
do_blocking_move_to_xy_z = { -87.47, -50.50, 5.00} fr_mm_s=0.00
_goto_tower_x
goto_tower @ 210.00
tower_vec (coefficients) = { -0.87, -0.50, }
_man_probe_pt = { -87.47, -50.50, }
do_blocking_move_to_xy_z = { -87.47, -50.50, 5.00} fr_mm_s=0.00
echo: cold extrusion prevented
```

KosVas91, 2019-12-02:
Almost the same problem on delta (FLSUN QQ-S) with MKS Robin Mini board. Press Calibrate X(/Y/Z) - printer just hangs and to not respond to touchscreen.

istyszy, 2019-12-02:
I have the same problem, 
Anycubic Kossel Linear Plus + SKR V1.3 + TMC2130 +2004 LCD.

Allso i have problem with delta calibration, G33
everytime the same result:  Correct delta settings with M665 and M666
I have no ideea that the problem is....
Could we modify the firmware for more detailed error?  (over serial) just a thought.

sjasonsmith, 2019-12-02:
Can anyone responding to this echoing the same problem explain whether these options are actually useful? I am not sure if they are needed, or if they are completely replaced by G33 delta calibration.

I don’t have time to look any further into this right now. I don’t actually use these options, so I wouldn’t even notice if they were gone.

istyszy, 2019-12-03:
who knows the last working version?

boelle, 2019-12-07:
@vijexa since 2.0 was just released a few days ago has this changed this issue at all?

nivaman, 2019-12-08:
Hi.

I can confirm, that build from 1.12.2019 still has this issue.

sjasonsmith, 2019-12-20:
This is happening because the planning buffer fills up and `_man_probe_pt` (in menu_delta_calibrate.cpp) is called recursively when the menu update is refreshed.

I've tested a hacked in a fix by using a static local to track whether a `_man_probe_pt` is already in progress, and simply returning when called recursively.

This doesn't seem like the right fix, but I'm having trouble figuring out how to avoid recursive callbacks from the menu when `idle()` is called (deep inside `_man_probe_pt`).

@thinkyhead, do you have a suggestion for this, or know who I could talk to for a recommendation?

boelle, 2020-01-10:
> @thinkyhead, do you have a suggestion for this, or know who I could talk to for a recommendation?

sjasonsmith, 2020-01-10:
@InsanityAutomation andbi have been talking about this a bit. The problem is that he doesn’t have a delta and I don’t presently have time.

We are currently considering whether these are completely obsolete. They have probably been broken for a long time, and if you use G33 I don’t think you shouldn’t need them. Even if you do need to manually calibrate, these don’t actually store any settings, they simply let you manually probe a height through the LCD, so they really are not that great of a solution.

One option is to simply delete them. This will prevent people who don’t actually need them from reporting issues. Nobody replied to my question on this issue asking whether this capability actually matters. If people complain that they are still important, they could be replaced with something more useful that actually does something, such as a G33 P3 to probe the tower heights.

boelle, 2020-01-24:
@vijexa when you get time please have a retest

and as bugfix 2.0.x is updated daily please retest say every 14 days or so

sjasonsmith, 2020-01-24:
My pull request mentioned above has not been merged yet. They are welcome to test with the changes in my PR, but I can guarantee it is still broken in bugfix-2.0.x.

sjasonsmith, 2020-01-26:
@vijexa @KosVas91 @istyszy @nivaman 

This fix was just merged into bugfix-2.0.x. Could one of you try it out and verify that it fixes the issue for you?

boelle, 2020-02-02:
@vijexa @KosVas91 @istyszy @nivaman

as @sjasonsmith just asked could one of you try it out and verify that it fixes the issue for you?

sjasonsmith, 2020-02-02:
@boelle I think you can go ahead and close it. They've had a week, so we'll just need to depend on my testing when I created the fix.

boelle, 2020-02-02:
oki, just ping me if we need to reopen, i just have a rule of thumb not to close verified ones

closed 2020-02-02 by boelle

nivaman, 2020-02-13:
I apologize if this is not aloved, but just for info - I have loaded firmware from end of January and can confirm that Delta calibration is now working.

InsanityAutomation, 2020-02-13:
@nivaman definitely allowed and appreciated! Thanks for the confirmation :)