MarlinFirmware/Marlin #16008: [BUG] No PWM on STM32F446 + marlin 2.0bugfix + bltouch

DropbearNinja, 2019-11-26:
### Description

Setting up marlin with bltouch on a stm32f446 board
The probe does its' self test, but there is no blue led (indicating a pwm signal), and does not respond to any gcode
The probe is fine, checked on another printer and functions properly
I've got it hooked up to the right PWM pin

### Steps to Reproduce

In my pins.h"
```
#define NUM_SERVOS 1
#define SERVO0_TIMER_NUM 8
#define SERVO0_PIN         PD15
```

In config.h
`#define BLTOUCH`

In config_adv.h
no extra options enabled

**Expected behavior:** 
PWM signal

**Actual behavior:** 
no PWM signal

#### Additional Information
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3891743/Marlin.zip)

Many thanks for help

xC0000005, 2019-11-27:
What environment and variant is set for this board?

Drachenkaetzchen, 2019-11-29:
I can kinda confirm the issue. In my case, it's a Fysetc S6 board. Toggling the pin is fine with `M43`, verified using an oscilloscope. No change to the voltage when using `M280 P0 S10´ (stays at 0V).

Could be a general PWM issue, as the fan mosfets on the S6 also don't seem to receive any signal (need to verify though).

boelle, 2019-11-29:
@DropbearNinja still arround?

xC0000005, 2019-11-29:
I would look at the variant’s pins configuration to see if PWM is configured for it. The ST core doesn’t log an error if it isn’t. It’s in the variant pins c file.

DropbearNinja, 2019-11-30:
Still here
Board is Rumba32
Environment?

boelle, 2019-11-30:
> Environment?

what do you use to compile marlin

DropbearNinja, 2019-11-30:
Arduino IDE

DropbearNinja, 2019-11-30:
For comparison, mk3duo works out of the box

boelle, 2019-11-30:
what version?

have you tried vscode+platformio ?

DropbearNinja, 2019-11-30:
1.8.10

No I haven't

boelle, 2019-11-30:
try 1.9 beta

that and vscode is the recommended these days

boelle, 2019-12-02:
@DropbearNinja did it make any difference?

DropbearNinja, 2019-12-03:
1.9 no
and i cant use vscode - platformio doesn't support my board

DropbearNinja, 2019-12-03:
So my bltouch works fine using mk4duo + stm32duino + rumba32 (stmf446) + arduino IDE

Changing just to Marlin, everything else the same, no luck - has to be some way in which Marlin uses servo.h/cpp or timers.h?

boelle, 2019-12-03:
i have no idea, i'm just the custodian dusting off the issue list and make sure title of issue is right  and that people dont forget to read the issue template

DropbearNinja, 2019-12-05:
I managed to solve this, see https://github.com/Aus3D/RUMBA32/issues/8

closed 2019-12-05 by DropbearNinja