MarlinFirmware/Marlin #16009: Get GCC ERRORs from Lint in PlatformIO with ATOM

GadgetAngel, 2019-11-26:
### Description

### Steps to Reproduce
**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files. 
links to my files:
https://drive.google.com/open?id=1U75R1UnMs2UsOXjmV9bdxrPTSJ9viGfe

1. download MarlineV2-bugfix
2.Make any change to Configuration.h or Configuration_adv.h
3. get GCC errors when compiling with PlatformIO

**Expected behavior:** [What you expect to happen]
now GCC errors should occur

**Actual behavior:** [What actually happens]
[image]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

boelle, 2019-11-26:
have you tried vscode+platformio ? i think atom is dead and outdated

ellensp, 2019-11-27:
These configuration.h files are not current...  Many define names have to be updated  
I suspect this is not from the current official marlin github repository.

boelle, 2019-11-27:
@GadgetAngel if these files are not from the marlin github repro we cant help you, you will have to ask for help where you got marlin from

GadgetAngel, 2019-11-27:
I got those configuration.h files from the Marlin branch about 3 months ago.  Are you telling me you want me to download and reset ALL my customized Marlin firmware again from scratch?? That’s a lot of work, don’t you think?  If The only way you will help me is to either switch to VScode+Platformio and re-Download the MarlinV2.0-bugfix AND redo All my customizations then I ask that you provide me the http link to download the version you want me to have.

GadgetAngel, 2019-11-27:
BTW, I thought ATOM+Platformio was the official editor for Marlin, is that not correct? Because that’s why I am using the ATOM editor.

boelle, 2019-11-27:
nope vscode + platformio is the recommended one

https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x

boelle, 2019-11-27:
if the customize work is in the 2 config files you can use a tool like winmerge to compare your config with the one in marlin and then copy over the settings

GadgetAngel, 2019-11-27:
since that is the case, please ask the people who maintain the pages for Marlin to Remove ATOM+Platformio as one of the editors to be used: 

[image]

I have been dealing with software all my life and I am a master degree EE who has done instrumentation and control engineering as my career. I am now retired. 

I will switch to VScode+Platformio but if you will not help people who use ATOM+Platformio then please remove it from the list of recommended editors. It’s no big deal for me to move to VScode but for other users this would be a big task. 

I am still of the opinion that these GCC code errors need to be solved for ATOM+Platformio because ATOM+Platformio IS Listed as an editor.  I hope you can determine what the problem is.  If the solution is to NOT use ATOM, fine just tell me that.

Sorry I am now stepping down off my soap box.

boelle, 2019-11-27:
> but if you will not help people who use ATOM+Platformio then

where did i say that?

p3p, 2019-11-27:
@GadgetAngel Atom is still a valid editor but PlatformIO decided to move to VSCode as their editor of choice, the editor you use is of course up to you.

Saying that, the errors you are seeing are a problem with the linter software not being able to parse the C Preprocessor correctly as it doesn't know to pull in the required headers, This isn't a problem with the Marlin build, it's a cosmetic feature of the editor, so there is nothing we can do about them.

GadgetAngel, 2019-11-27:
> > but if you will not help people who use ATOM+Platformio then
> 
> where did i say that?

If I read it wrong I apologize to you

GadgetAngel, 2019-11-27:
> @GadgetAngel Atom is still a valid editor but PlatformIO decided to move to VSCode as their editor of choice, the editor you use is of course up to you.
>
> Saying that, the errors you are seeing are a [...]

Thank you p3p!! So basically if I stick with ATOM editor I will always see this due to the linter software....OK so I am now going to make the switch to VScode...I already use VScode for openHAB so it’s no big deal... You can close this bug so far as I am concern

Thank you all for your input

But mostly for maintaining Marlin SW

If I offended anyone, I do apologize

boelle, 2019-11-27:
> nope vscode + platformio is the recommended one

but as @p3p said its up to you what you use :-D

but yes there are some benefits to vscode and platformio combo

closed 2019-11-27 by GadgetAngel

boelle, 2019-11-27:
> If I offended anyone, I do apologize

not me, passed the age of 42 and found out that life is short enough as is, so why waste time on small things :-D

Patag, 2019-11-27:
Wow I'm 10y older than you !?! @boelle