MarlinFirmware/Marlin #16010: [BUG] (short description)

GurenWolf, 2019-11-26:
**Environment**

3D Pirnter: Anycubic Kossel Linear Plus
Board: Bigtreetech SKR V1.3
Stepper Drivers: TMC2208 V3
Firmware: Marlin Bugfix 2.0.x (November 26th, 2019 - Issue #15994 proved and fixed on this trial)
Probe: Anycubic provided probe (V2 on firmware) 

**Description of the bug**

While testing UBL, when doing the step by step UBL, right after running step 3 on Validate Mesh, this message appears:

Unknown command: "G26 C B60 H200 P"

**Steps to Reproduce**

1. Run step 1 and step 2 of UBL without auto homing.
2. Run step 3 (Validate Mesh)

**Expected behavior:**

;---------------------------------------------
;--- Fine Tuning of the mesh happens below ---
;---------------------------------------------
G26 C P5.0 F3.0 ; Produce mesh validation pattern with primed nozzle (5mm) and filament diameter 3mm
; PLA temperatures are assumed unless you specify, e.g., B 105 H 225 for ABS Plastic

**Actual behavior:**

Unknown command: "G26 C B60 H200 P"

tatusah, 2019-11-26:
Looking at your configuration it seems that you haven't enabled the the mesh validation feature. Uncomment the line with //#define G26_MESH_VALIDATION to enable it.

GurenWolf, 2019-11-26:
@tatusah, thanks a lot for this, I thought it was enabled by default and therefore assumed that it was a coding error.

closed 2019-11-26 by GurenWolf