MarlinFirmware/Marlin #16011: [BUG] Latest UI related commits broke serial com with BTT TFT35 V3.0

Patag, 2019-11-26:
Since commits from 24 November, I can't get communication with TFT35v3.0 anymore when used in touchscreen mode (all is OK in 12864 mode). The LCD module displays "No connected printer" indefinitely. I had to revert all the UI related commits to get correct operation back.
My conf : SKR Pro / TMC 5160 x 6 / BBT TFT35 v3 (up to date).
Even if they won't be useful and before @boelle ask them and close the issue ;-), find attached my conf files :
[Config.zip](https://github.com/MarlinFirmware/Marlin/files/3893504/Config.zip)
Thanks in advance

boelle, 2019-11-26:
not closing anything this new :-D

Patag, 2019-11-26:
Thanks @boelle. I still have to make progress with issue titles...

boelle, 2019-11-26:
hehe, i added that yesterday so we quickly can see what version of marlin it is
if people use old version then we can start by asking them to use 2.0.x as that is the only one updated

Patag, 2019-11-26:
Smart change. I'll try to remember it !

sjasonsmith, 2019-11-26:
@Patag, can you identify a specific commit that causes the failure?

Patag, 2019-11-26:
Hum, not able to identify the commit Id , but I can narrow down to the ones from 24 november and after

boelle, 2019-11-26:
what if you take the last on 23 november? does that work? if so take next commit until it fails

thisiskeithb, 2019-11-27:
I’m running a BigTreeTech TFT35 V3 on an SKR 1.3 and it’s been working great without any hiccups. I wonder why it doesn’t work on the Pro.

Did you update the TFT to the [latest firmware](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware)?

Patag, 2019-11-27:
@boelle 23 nowember works, 24 november and after don't.
@thisiskeithb yes

Patag, 2019-11-27:
I've probably messed up my local repository. I just re-based it and all seem to be OK.
Sorry for the inconvenience.

closed 2019-11-27 by Patag