MarlinFirmware/Marlin #16012: Z probe offset not working

thebest07111, 2019-11-26:
hello,

My z probe offset worked one time. I changed it and the nozzle went down. I stored the settings and restarted the printer. That it was at 0 again. Now whenver I move my nozzle down with the Z probe offset it wont do anything at all I can move it to -10 but it just wont go down. Anyone knows why. I have the E3 mini.

Here are my firmware files:

https://we.tl/t-oZUjBnqPvx

boelle, 2019-11-26:
attach your config files as zip instead

just make a zip file and edit the first post, then drop the zip file while you edit and it will attached

boelle, 2019-11-26:
you do know that eeprom comes before settings, so you need to do M502 then M500 to reset eeprom

thebest07111, 2019-11-26:
> you do know that eeprom comes before settings, so you need to do M502 then M500 to reset eeprom

I did so this is the output:
```
echo:V72 stored settings retrieved (604 bytes; crc 3900)
echo: G21 ; Units in mm (mm)
echo: M149 C ; Units in Celsius
echo:Filament settings: Disabled
echo: M200 D1.75
echo: M200 D0
echo:Steps per unit:
echo: M92 X80.00 Y80.00 Z400.00 E93.00
echo:Maximum feedrates (units/s):
echo: M203 X500.00 Y500.00 Z5.00 E25.00
echo:Maximum Acceleration (units/s2):
echo: M201 X500.00 Y500.00 Z100.00 E5000.00
echo:Acceleration (units/s2): P R T
echo: M204 P500.00 R500.00 T500.00
echo:Advanced: B S T J
echo: M205 B20000.00 S0.00 T0.00 J0.08
echo:Home offset:
echo: M206 X0.00 Y0.00 Z0.00
echo:Auto Bed Leveling:
echo: M420 S0 Z0.00
echo:Material heatup parameters:
echo: M145 S0 H185 B45 F255
echo: M145 S1 H240 B110 F255
echo:PID settings:
echo: M301 P21.73 I1.54 D76.55
echo:Z-Probe Offset (mm):
echo: M851 X-44.00 Y-14.00 Z0.00
echo:Stepper driver current:
echo: M906 X580 Y580 Z580
echo: M906 T0 E650
echo:Driver stepping mode:
echo: M569 S1 X Y Z
echo: M569 S1 T0 E
ok
```

It just wont move down when changing the z probe offset.

tatusah, 2019-11-26:
Uncomment //#define BABYSTEP_ZPROBE_OFFSET in configuration adv to usethe motorized offset editing after homing and moving the nozzle to z 0. Then you can close this issue, this is more of a configuration problem and preferably handled in the discord channel

boelle, 2019-11-26:
Marlin Discord server. Join link: https://discord.gg/n5NJ59y

yep, support and questions go there, here is for bugs only (and feature requests

closed 2019-11-26 by boelle