MarlinFirmware/Marlin #16015: [BUG] [Bugfix2.0.x] M112 in combination with Emergency_Parser does not behave properly

basgoossen, 2019-11-26:
With emergency parser enabled, M112 does not directly kill the machine. Instead it will pause for a second and afterwards moves from the buffer will be execured untill the buffer is empty.

This might lead to dangerous situations especially on big machines with potential long moves and large consequence.

More info and discussion about the problem can be found here: 
https://forum.v1engineering.com/t/skr-1-3-marlin-2-0-m112-em-halt-issue/12945/10

boelle, 2019-11-26:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

basgoossen, 2019-11-26:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3894213/Configuration.zip)
As per request. However my personal configuration files here are seemingly irrelevant. As the bug can be reproduced in 2 simple steps:

- Enable EMERGENCY_PARSER
- Send:
     G1 X50 Y50 F500
     G1 X100 Y50 F500
     M112
Next: Watch the machine complete the GCODE instead of EM Stopping.

boelle, 2019-12-07:
since 2.0 was just released a few days ago has this changed this issue at all?

boelle, 2020-01-07:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-01-07 by boelle