MarlinFirmware/Marlin #16025: [BUG] (short description)

Gerggg, 2019-11-27:
### Description

Hello,

I think I found a bug, when I turn on the function HYBRID_THRESHOLD and when printing the speed passes the limit configured, I have a shift of layers.
If I change the values so that the speed of the motors stays below the limits, the offset of layers disappears

### Steps to Reproduce

1- activate the HYBRID_THRESHOLD
2- start printing with speeds move limits configured in Configuration_adv.h
3- wait for the speeds to be shifted and notice the shifting of layers.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3896208/Marlin.zip)
I specify that I disabled the HYBRID_THRESHOLD in Configuration_adv.h to solve my problem

#### Additional Information

My config
SKR 1.3
TMC2209 UART

Gerggg, 2019-11-29:
Ok, thanks for the info but I prefer to disable the option.

boelle, 2019-11-29:
@Gerggg so there is no issue after all?

Gerggg, 2019-11-30:
I do not know how to say it because I have some shifts when I use the function whereas I did not have any before.

boelle, 2019-12-07:
since 2.0 was just released a few days ago has this changed this issue at all?

boelle, 2019-12-24:
btw, who can confirm there is an issue? use the same configs as OP and the same hardware if possible

rado79, 2019-12-26:
confirm - see https://github.com/MarlinFirmware/Marlin/issues/16329

boelle, 2020-01-07:
so #16329 is just an dublicate of this one?

rado79, 2020-01-07:
> so #16329 is just an dublicate of this one?

Looks like the same bug. Should i close #16329 ? Maybe there are some more findings.

tatusah, 2020-01-07:
> For example, StealthChop operation is used for precise low speed positioning, while
> SpreadCycle shall be used for highly dynamic motion. TPWMTHRS determines the transition velocity.
> Read out TSTEP when moving at the desired velocity and program the resulting value to TPWMTHRS.
> Use a low transfer velocity to avoid a jerk at the switching point.
> A jerk occurs when switching at higher velocities, because the back-EMF of the motor (which rises
> with the velocity) causes a phase shift of up to 90° between motor voltage and motor current. So
> when switching at higher velocities between voltage PWM and current PWM mode, this jerk will occur
> with increased intensity. A high jerk may even produce a temporary overcurrent condition (depending
> on the motor coil resistance). At low velocities (e.g. 1 to a few 10 RPM), it can be completely
> neglected for most motors. Therefore, consider the switching jerk when choosing TPWMTHRS. Set
> TPWMTHRS zero if you want to work with StealthChop only.

The layer shift might not only be caused by acceleration. Based on [Trinamic documentation](https://www.trinamic.com/fileadmin/assets/Products/ICs_Documents/TMC2209_Datasheet_V103.pdf) there probably always is some jerk happening with high speed hybrid threshold. Jerk doesn't matter in some applications but in 3D printing where predictability is a necessity it's a bit different. Also many seem to want to optimize the decibels even when cooling fans might make more noise. I guess it's up to each own preference how to optimize the drivers depending on wheter one prefers quiet operation or something else. Trinamic seems to suggest low threshold veloticy of 1 to few 10 RPM, which on my system would mean maximum speed of around 6 mm/s for the 10 RPM So basically it would then be always in Spreadcycle mode. I don't think SpreadCycle is that much louder when you count in all the other noises.  Yet, after some tests with Hybrid Threshold I have personally ended up using just the StealthChop mode.

If I ever seem a need for hybrid threshold I might then start with low threshold speed and continue raising it until I start to see effects of jerk and possible layer shift.

boelle, 2020-01-07:
> Looks like the same bug. Should i close #16329 ? Maybe there are some more findings.

yes if its the same bug the one created first should stay
we only need one bug report for the same issue

boelle, 2020-01-18:
@tatusah so this is more a problem if the user sets the threshold to high and not so much a problem with the marlin code?

tatusah, 2020-01-18:
I would say so, this is among the many features that should be tuned specific to everyone's own system and expectations. I think setting the threshold is a bit like timing the gear shifts on a manual transmission car there's a certain sweet spot.

rado79, 2020-01-19:
> @tatusah so this is more a problem if the user sets the threshold to high and not so much a problem with the marlin code?

Problem here on threshold 80mm/s or 250mm/s or everywhere between
Or do you mean threshold should be set lower?  There is no need for this option in lower speeds i mean.

Btw i do not need spreadcycle or hybrid, my CoreXY does stealthchop pretty well until 250mm/s,
but I noticed while doing calibrations.
And on CoreXY threshold does not refer to x and y direction speed, but to steppermotor 1 and 2 rotation speed (diagonal move - I don't know how I could better describe it in english)

boelle, 2020-01-23:
closing as this seems to be a config issue and not a bug

also the OP seems to have figured his issue since he is not replying anymore

closed 2020-01-23 by boelle