MarlinFirmware/Marlin #16028: G34 not working with latest release

davidcgu, 2019-11-27:
Good afternoon

With latest release I can not make work auto-align with dual Z.

My bed size is 310'x310
Bltouch installed at NOZZLE_TO_PROBE_OFFSET { 45, 0, -2.95 }
MIN_PROBE_EDGE 10
AUTO align enabled with positions defined as follows:

define Z_STEPPER_AUTO_ALIGN
#if ENABLED(Z_STEPPER_AUTO_ALIGN)
  // Define probe X and Y positions for Z1, Z2 [, Z3]
  #define Z_STEPPER_ALIGN_XY { {  45, 150 }, { 280,  150 } }

The definitions are a bit different compared on how was done before but I set same values(actuallly using version downoaded on 1 Nov and working fine there).

When running G34 from console just returns me "cancelled" and does not work.

Any idea? 

Thanks in advance

sjasonsmith, 2019-11-27:
Can you upload your config files?

davidcgu, 2019-11-27:
Here We go   https://drive.google.com/open?id=1mXYkwJYa34DSCnQIU3o03ejxh6aKnUzv

sjasonsmith, 2019-11-27:
The first thing I would try, is to move your probe points further to the middle of the bed. The positions looks like they should be probable, but this will identify if Marlin is having issues probing right at the edge of the probable boundary.
`#define Z_STEPPER_ALIGN_XY { { 65, 150 }, { 245, 150 } }`

Next, enable DEBUG_LEVELING_FEATURE and provide us with the output from the console when the error occurs. You will need to run `M111 S32` prior to `G34` to enable additional output.
```
M111 S32
G34
```

Also, can you confirm the date (or even better, commit number) your Marlin source is from? An issue was fixed in G34 on Nov 19th, so I assume it was working for other people at that time. At least one other person confirmed that fixed issues for them.

Unfortunately my printer I use G34 with is not at my home, so I cannot readily test it out on my machine.

davidcgu, 2019-11-27:
Good afternoon

I will give a try and provide the info, Marlin relase downloaded at 25 Nov, just two days ago.

Apparently behaviour is the same than previous version when you define some position out of limits considering max-min travel limts, edge and sensor offset  , however the ones I have now I'm comletly sure are fine.

Regards

sjasonsmith, 2019-11-27:
Your values look ok to me as well, but there have been some probing changes recently so it is possible something is not working correctly.

davidcgu, 2019-11-27:
Tested: with X55-290 G34 completes however on right side sure the test is not performed at 290mm while is far away from it I measure with a ruler and seems to be doing it at X=200, don't ask me why...

If I move manually  with G1 X0 Y150 F5000 or G1 X310 Y150 F5000 then it moves perfectly well from one border to the other of the bed.

Regards

davidcgu, 2019-11-28:
Test completed!!

1- test points at X65-240  G34 succesfully completed.
2- test points at X55-290 G34 succesfully completed.

For me apparenlty seems to be adding the edge on top of the Z probe offset, as I have the bl touch at X+45 from nozzle and the edge defined at 10 mm what makes now it's able to complete with X at 55 mm on left side.

So I guess this calculation is done opposite than should be on Marlin, looks a bug.

Thanks for your time and regards.

sjasonsmith, 2019-11-28:
Would you mind trying again, using X=46? I don't immediately see anything in the code which would cause the offset and minimum probe edge to be added together. This will help us know whether it is _exactly_ 45 that doesn't work, or if it actually is MIN_PROBE_EDGE.

You should be able to reconfigure that point without rebuilding the firmware, by sending the following command:

`M422 S1 X46`

If you execute `M422` with no arguments, you can verify that the correct coordinates are active.

davidcgu, 2019-11-28:
👍  Doing like this it works also at X45:

Send: M422
Recv: M422 S1 X46.00 Y150.00
Recv: M422 S2 X290.00 Y150.00

I don't know why I can ensure you flashing with this straight on config advance it doesn't work I tried several times.

Regards

sjasonsmith, 2019-11-28:
Maybe something is stored in your EEPROM that is messings things up. Perhaps use M502 then M500 to reset your EEPROM contents to defaults.

The alignment probing locations are not stored in the EEPROM, but maybe there is something else funny in there.

If you still have issues, can you check M422 right after power-on, to see if it says something unexpected?

EvilGremlin, 2019-11-29:
Yep, as of 29.11.19 it's broken:
[config.zip](https://github.com/MarlinFirmware/Marlin/files/3906359/config.zip)
`Recv: ok
[...]
Send: M502
Recv: ok
[...]
Send: M500
Recv: ok
[...]
Send: G28
[...]
Recv: echo:busy: processing
Printer seems to support the busy protocol, will adjust timeouts and set busy interval accordingly
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: X:80.00 Y:48.00 Z:0.00 E:0.00 Count X:12800 Y:7680 Z:0
Recv: ok
[...]
Send: M113 S29
Recv: ok
[...]
Send: G34
Recv: 
Recv: ITERATION: 1
Recv: Probing failed.
Recv: G34 aborted.
Recv: ok`

EvilGremlin, 2019-11-29:
Here's M422 behavoiur log:
X10 is actually unreachable, nevertheless behaviour of G34 is not consistent.
> [...]
> Send: M422
> Recv: M422 S1 X10.00 Y100.00
> Recv: M422 S2 X190.00 Y100.00
> Recv: ok
> [...]
> Send: M422 S1 X30 Y100
> Recv: ok
> [...]
> Send: M422 S2 X170 Y100
> Recv: ok
> [...]
> Send: G34
> [...]
> Recv: X:80.00 Y:48.00 Z:0.00 E:0.00 Count X:12800 Y:7680 Z:0
> Recv: 
> Recv: ITERATION: 1
> [...]
> Recv: echo:busy: processing
> [...]
> Recv: 
> Recv: DIFFERENCE Z1-Z2=0.41
> Recv: 
> Recv: ITERATION: 2
> [...]
> Recv: echo:busy: processing
> [...]
> Recv: 
> Recv: DIFFERENCE Z1-Z2=0.27
> Recv: 
> Recv: ITERATION: 3
> [...]
> Recv: echo:busy: processing
> [...]
> Recv: 
> Recv: DIFFERENCE Z1-Z2=0.02
> Recv: Target accuracy achieved.
> Recv: Did 3 iterations of 3
> Recv: Accuracy: 0.02
> [...]
> Recv: echo:busy: processing
> [...]
> Recv: X:80.00 Y:48.00 Z:0.00 E:0.00 Count X:12800 Y:7680 Z:0
> Recv: ok
> [...]
> Send: M422 S1 X10 Y100
> Recv: ?(X) out of bounds.
> Recv: ok
> [...]
> Send: M422 S2 X190 Y100
> Recv: ok
> [...]
> Send: G34
> Recv: 
> Recv: ITERATION: 1
> [...]
> Recv: echo:busy: processing
> [...]
> Recv: 
> Recv: DIFFERENCE Z1-Z2=0.01
> Recv: Target accuracy achieved.
> Recv: Did 1 iterations of 3
> Recv: Accuracy: 0.01
> [...]
> Recv: X:80.00 Y:48.00 Z:0.00 E:0.00 Count X:12800 Y:7680 Z:0
> Recv: ok

boelle, 2019-12-07:
since 2.0 was just released a few days ago has this changed this issue at all?

boelle, 2019-12-24:
btw, who can confirm there is an issue? use the same configs as OP and the same hardware if possible

davidcgu, 2019-12-24:
Runing with latest bugfix from yesterday works well, just do a factory reset  Eeprom after flashing.

boelle, 2019-12-24:
yep should always do factory reset after flashing, so this one is solved 100% ?

davidcgu, 2019-12-24:
For me seems definetly closed, no more issues with latest release, I think it can be closed!  Thanks  alot guys and merry Xmas for everybody!

boelle, 2019-12-24:
i just had to ask so i dont step on any toes :-D

but will close and also merry xmas to everyone

should someone else have the issue if using the same config then we can reopen

closed 2019-12-24 by boelle