MarlinFirmware/Marlin #16029: [BUG] Can't Compile (STM32F4-RUMBA32) fails SPI / DMA

BoHoehne, 2019-11-27:
### Description
Trying to compile Marlin 2.x for RUMBA32 (STM32F4).
The stepper drivers are TMC5160 with SPI connection.
Environment: VSCode / PlatformIO

Error:
Marlin\src\HAL\HAL_STM32_F4_F7\HAL_SPI.cpp:125:9: error: 'class SPIClass' has no member named 'dmaTransfer'; did you mean 'transfer'?
     SPI.dmaTransfer(0, const_cast<uint8_t*>(buf), nbyte);
         ^~~~~~~~~~~
         transfer
Marlin\src\HAL\HAL_STM32_F4_F7\HAL_SPI.cpp: In function 'void spiSendBlock(uint8_t, const uint8_t*)':
Marlin\src\HAL\HAL_STM32_F4_F7\HAL_SPI.cpp:158:9: error: 'class SPIClass' has no member named 'dmaSend'
     SPI.dmaSend(const_cast<uint8_t*>(buf), 512);
         ^~~~~~~ 

1. Working VSCode with PlatformIO 
2. Settings in platform.ini: -> [platformio] -> default_envs = STM32F4
3. Configured  "configuration.h" as follows:

#define MOTHERBOARD BOARD_RUMBA32
..
#define X_DRIVER_TYPE  TMC5160
#define Y_DRIVER_TYPE  TMC5160
#define Z_DRIVER_TYPE  TMC5160
#define E0_DRIVER_TYPE TMC5160
#define E1_DRIVER_TYPE TMC5160
#define E2_DRIVER_TYPE TMC5160

**Expected behavior:** 
Successful compilation

**Actual behavior:** 
Compile aborts with above errors.

sjasonsmith, 2019-11-27:
The STM32 HALs get pretty confusing.

I _believe_ that when you build for a Rumba32 inside Arduino IDE it uses HAL_STM32, whereas that STM32F4 environment uses HAL_STM32_F4_F7.

The SKR Pro is the best example of an STM32F4 building with HAL_STM32. The FYSETC S6 does also, but it was just added yesterday and isn't fully tested.

There will likely be more work needed than simply copying and editing a new environment, but that is not a process I am familiar with.

LinoBarreca, 2019-12-06:
just retarget the board to stm32 hal

boelle, 2019-12-07:
@BoHoehne did you figure the issue?

BoHoehne, 2019-12-07:
When I set the board in platformio.ini as "STM32_hal" I get the following error:

Error: BoardConfig: Board is not defined

Unsure what to configure:
[env:STM32_hal]
board = ???

Unsure what to insert there. 
Hints are welcome.

LinoBarreca, 2019-12-08:
@BoHoehne I don't know how to build rumba but to change the hal change 
```
[env:STM32F4]
platform      = ststm32
framework     = arduino
board         = disco_f407vg
build_flags   = ${common.build_flags} -DUSE_STM32GENERIC -DSTM32GENERIC -DSTM32F4 -DMENU_USB_SERIAL -DMENU_SERIAL=SerialUSB -DHAL_IWDG_MODULE_ENABLED
lib_deps      = ${common.lib_deps}
lib_ignore    = Adafruit NeoPixel, TMCStepper
src_filter    = ${common.default_src_filter} +<src/HAL/HAL_STM32_F4_F7> -<src/HAL/HAL_STM32_F4_F7/STM32F7>
monitor_speed = 250000
```
into
```
[env:STM32F4]
platform      = ststm32
framework     = arduino
board         = disco_f407vg
build_flags   = ${common.build_flags}  -DTARGET_STM32F4 -DSTM32F4 -DMENU_USB_SERIAL -DMENU_SERIAL=SerialUSB
  -IMarlin/src/HAL/HAL_STM32
lib_deps      = ${common.lib_deps}
lib_ignore    = Adafruit NeoPixel, TMCStepper
src_filter    = ${common.default_src_filter} +<src/HAL/HAL_STM32>
monitor_speed = 250000
```

BoHoehne, 2019-12-09:
Thank you.
That changed something. Now the error looks like this:

Marlin\src\HAL\shared\backtrace\../../../core/../inc/../HAL/./HAL_STM32/HAL.h:54:21: error: 'SerialUSB' was not declared in this scope

wap32, 2019-12-10:
Check Aus3D/RUMBA32#11, I posted my platformio config which works for me.

BoHoehne, 2019-12-11:
@wap32 <- Thank you so much !!!

Compile went thru without any problem. Yippie !!

boelle, 2019-12-11:
will close as this is most likely a config problem and not a marlin problem

closed 2019-12-11 by boelle