MarlinFirmware/Marlin #16030: [BUG] SKR E3/mini/DIP Motherboard Incorrectly Evaluated in "timers.h" for HAL_STM32F1

noahdavis319, 2019-11-27:
### Description

The motherboard is incorrectly evaluated in timers.h when determining the SERVO0_TIMER_NUM. The "MB" macro is returning false when defined as "BTT_SKR_MINI_E3_V1_0", or "BIGTREE_SKR_E3_DIP", or "BTT_SKR_MINI_E3_V1_2", or "MKS_ROBIN_LITE".

### Steps to Reproduce

Marlin bugfix-2.0.x (pulled down 11/26/2019)
Ubuntu 18.04 LTS.
Visual Studio Code (latest)
PlatformIO (Home 3.0.1, Core 4.1.0)

VSCode provides real-time configuration-based syntax highlighting as you #define and un-#define various configuration options. When viewing HAL/HAL_STM31F1/timers.h lines 73-82 are incorrectly highlighted which leads me to believe that the "MB" macro is not correctly evaluating which motherboard is being used.

No Configuration.h or Configuration_adv.h attached as the only change needed is to `#define MOTHERBOARD BOARD_BIGTREE_SKR_E3_DIP` in configuration.h

**Expected behavior:**

Since the STM31F1 is a HIGH_DENSITY board, then line 76 of timers.h (`#define SERVO_TIMER_NUM 8`) should be lit.

**Actual behavior:** [What actually happens]

Line 73 (the motherboard evaluation expression using the MB macro) returns false and line 81 becomes lit (which still evaluates SERVO0_TIMER_NUM to 8).

#### Additional Information

This bug was found in an effort to resolve any possible timer issues such that BLTouch can work on SKR Mini E3/DIP/V1.2 using the PROBE and SERVO pins.

sjasonsmith, 2019-11-27:
That statement is working fine for my DIP build.

Syntax highlighting is not always perfect when it comes to preprocessor macros. To verify this is a highlighting issue and not a code issue, you could insert from `#error` statements like this, and see which is hit when you compile.

```
#if MB(BTT_SKR_MINI_E3_V1_0, BIGTREE_SKR_E3_DIP, BTT_SKR_MINI_E3_V1_2, MKS_ROBIN_LITE)
  // SKR Mini E3 boards use PA8 as FAN_PIN, so TIMER 1 is used for Fan PWM.
  #ifdef STM32_HIGH_DENSITY
    #error "Timer 8"
    #define SERVO0_TIMER_NUM 8  // tone.cpp uses Timer 4
  #else
    #error "Timer 3"
    #define SERVO0_TIMER_NUM 3  // tone.cpp uses Timer 8
  #endif
#else
  #error "Timer 1"
  #define SERVO0_TIMER_NUM 1  // SERVO0 or BLTOUCH
#endif
```

noahdavis319, 2019-11-27:
Good to know. I will close this. Thank you.

closed 2019-11-27 by noahdavis319

sjasonsmith, 2019-11-27:
I find that sometimes the syntax highlighting improves after you perform a build.

noahdavis319, 2019-11-27:
Perhaps you may be able to help since you have a DIP as well, are you able to get the BLTouch working using the SERVO and PROBE connector? Or is there still some work that needs to be done for expected functionality?

sjasonsmith, 2019-11-27:
I use servo and Xmin. I tried getting it work on the probe connector, but if you are using the probe as your end stop (as most of us do) I think Marlin expects it on Xmin.

You can edit the pins file to make it use the physical probe connector, but I don’t see any point. They are electrically the same.