MarlinFirmware/Marlin #16032: MIN_PROBE_EDGE ignored by G33 on Delta

eford321, 2019-11-27:
Description:

With latest bug fix-2.0.X (commit 5a98c83) MIN_PROBE_EDGE is ignored during probing with G33 calibration on Delta printers.  

Expected behavior:

MIN_PROBE_EDGE should move probing away from edges of the by the configured distance.

Actual behavior:

MIN_PROBE_EDGE has no effect. Tested with values of MIN_PROBE_EDGE = 10, 20 and 50.  No change to probing location. Issues M502 and M500 prior to each test.
Probing occurs at the DELTA_PRINTABLE_RADIUS plus NOZZLE_TO_PROBE_OFFSET causing the effector to move outside of bed.   

Thanks for any help!
[marlin configs.zip](https://github.com/MarlinFirmware/Marlin/files/3898081/marlin.configs.zip)

sjasonsmith, 2019-11-27:
I'll go try it out. I recently made changes to the G33 code. While MIN_PROBE_EDGE should be part of the equation I may not have specifically tested changing it.

sjasonsmith, 2019-11-27:
Are you sure you are running that commit you mentioned? I just tested both MIN_PROBE_EDGE (using 0, 10, and 40) and NOZZLE_TO_PROBE_OFFSET (using your offset values), and they all move probing toward the middle of the bed an appropriate distance.

The behavior you are describing sounds more like the behavior _before_ all of my delta probing changes. The probe offset was ignored during G33, causing issues like those you mention. Your configuration files look updated with my changes, but maybe you accidentally have some old source mixed in.

This is the code currently used to determine the probing radius. As you can see, they only reduce the calibration radius from the printable radius, it never increases it.

```
float delta_calibration_radius() {
  return FLOOR((DELTA_PRINTABLE_RADIUS - (
    #if HAS_BED_PROBE
      _MAX(HYPOT(probe_offset.x, probe_offset.y), MIN_PROBE_EDGE)
    #else
      MIN_PROBE_EDGE
    #endif
  )) * calibration_radius_factor);
}
```

eford321, 2019-11-27:
downloaded and built again. Working fine now. I must have mixed up my sources in Visual Studio. 

I'm sorry for wasting your time!

closed 2019-11-27 by eford321

sjasonsmith, 2019-11-27:
I'm just glad it is working, and not another urgent fix needed due to problems I caused :)