MarlinFirmware/Marlin #16034: TMC CONNECTION ERROR SKR pro v1.1 TMC2209

DJDOUBLEG, 2019-11-27:
[platformio]
src_dir      = Marlin
boards_dir   = buildroot/share/PlatformIO/boards
default_envs = BIGTREE_SKR_PRO

[common]
default_src_filter = +<src/*> -<src/config> -<src/HAL> +<src/HAL/shared>
extra_scripts = pre:buildroot/share/PlatformIO/scripts/common-cxxflags.py
build_flags = -fmax-errors=5 -g -D__MARLIN_FIRMWARE__
lib_deps =
  U8glib-HAL=https://github.com/MarlinFirmware/U8glib-HAL/archive/bugfix.zip
  LiquidCrystal@1.3.4
  TMCStepper@>=0.5.2,<1.0.0
  Adafruit NeoPixel@1.2.5
  Adafruit_MAX31865=https://github.com/adafruit/Adafruit_MAX31865/archive/master.zip
  LiquidTWI2=https://github.com/lincomatic/LiquidTWI2/archive/master.zip
  Arduino-L6470=https://github.com/ameyer/Arduino-L6470/archive/dev.zip
  SailfishLCD=https://github.com/mikeshub/SailfishLCD/archive/master.zip
  SailfishRGB_LED=https://github.com/mikeshub/SailfishRGB_LED/archive/master.zip
  SlowSoftI2CMaster=https://github.com/mikeshub/SlowSoftI2CMaster/archive/master.zip

#
# Bigtreetech SKR Pro (STM32F407ZGT6 ARM Cortex-M4)
#
[env:BIGTREE_SKR_PRO]
platform          = ststm32@>=5.7.0
framework         = arduino
platform_packages = framework-arduinoststm32@>=3.10700.191028
board             = BigTree_SKR_Pro
extra_scripts     = pre:buildroot/share/PlatformIO/scripts/generic_create_variant.py
build_flags       = ${common.build_flags}
  -DUSBCON -DUSBD_USE_CDC -DUSBD_VID=0x0483 -DUSB_PRODUCT=\"STM32F407ZG\"
  -DTARGET_STM32F4 -DSTM32F407_5ZX -DVECT_TAB_OFFSET=0x8000 -DHAVE_HWSERIAL6 -IMarlin/src/HAL/HAL_STM32
lib_deps          =
  U8glib-HAL=https://github.com/MarlinFirmware/U8glib-HAL/archive/bugfix.zip
  LiquidCrystal
  TMCStepper@>=0.5.2,<1.0.0
  Adafruit NeoPixel
  LiquidTWI2=https://github.com/lincomatic/LiquidTWI2/archive/master.zip
  Arduino-L6470=https://github.com/ameyer/Arduino-L6470/archive/dev.zip
lib_ignore        = SoftwareSerial, SoftwareSerialM
src_filter        = ${common.default_src_filter} +<src/HAL/HAL_STM32>
monitor_speed     = 250000

Processing BIGTREE_SKR_PRO (platform: ststm32@>=5.7.0; framework: arduino; board: BigTree_SKR_Pro)
-------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/BigTree_SKR_Pro.html
PLATFORM: ST STM32 5.7.0 > STM32F407ZG (192k RAM. 1024k Flash)
HARDWARE: STM32F407ZGT6 168MHz, 192KB RAM, 1MB Flash
DEBUG: Current (jlink) External (jlink, stlink)
PACKAGES: framework-arduinoststm32 3.10700.191028 (1.7.0), toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 13 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <U8glib-HAL> 0.4.1
|   |-- <SPI> 1.0
|   |-- <Wire> 1.0
|-- <LiquidCrystal> 1.5.0
|   |-- <Wire> 1.0
|-- <TMCStepper> 0.6.1
|   |-- <SPI> 1.0
|-- <Adafruit NeoPixel> 1.3.2
|-- <LiquidTWI2>
|   |-- <Wire> 1.0
|-- <Arduino-L6470> 0.7.0
|-- <IWatchdog> 1.0.0
|-- <EEPROM> 2.0.1
|-- <SPI> 1.0
|-- <Wire> 1.0
|-- <Servo> 1.1.2
Building in release mode

>>> m122
SENDING:M122
		X
Address		3
Enabled		false
Set current	800
RMS current	1436
MAX current	2025
Run current	25/31
Hold current	12/31
CS actual	0/31
PWM scale	0
vsense		0=.325
stealthChop	false
msteps		256
tstep		0
pwm
threshold
[mm/s]
OT prewarn	false
OT prewarn has
been triggered	false
off time	0
blank time	16
hysteresis
-end		-3
-start		1
Stallguard thrs	0
DRVSTATUS	X
stst
olb
ola
s2gb
s2ga
otpw
ot
157C
150C
143C
120C
s2vsa
s2vsb
Driver registers:
		X	0x00:00:00:00	 Bad response!
Testing X connection... Error: All LOW

# NO SUPPORT REQUESTS PLEASE

Support Requests posted here will be automatically closed!

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. See https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question.

For best results getting help with configuration and troubleshooting, please use the following resources:

- RepRap.org Marlin Forum http://forums.reprap.org/list.php?415
- Tom's 3D Forums https://discuss.toms3d.org/
- Facebook Group "Marlin Firmware" https://www.facebook.com/groups/1049718498464482/
- Facebook Group "Marlin Firmware for 3D Printers" https://www.facebook.com/groups/3Dtechtalk/
- Marlin Configuration https://www.youtube.com/results?search_query=marlin+configuration on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a Bug Report at https://github.com/MarlinFirmware/Marlin/issues/new/choose).

boelle, 2019-11-27:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

shitcreek, 2019-11-27:
Have you searched through the issue queue to see if this issue hasn't been brought up and resolved?
Such as these: https://github.com/MarlinFirmware/Marlin/issues/15415, https://github.com/MarlinFirmware/Marlin/issues/15821, https://github.com/MarlinFirmware/Marlin/issues/15840

boelle, 2019-11-27:
@shitcreek so this is an dublicate?

DJDOUBLEG, 2019-11-27:
[conf.zip](https://github.com/MarlinFirmware/Marlin/files/3898616/conf.zip)
Conf zipped

boelle, 2019-11-27:
check these issues and see if they are not the same
#15415, #15821, #15840

sjasonsmith, 2019-11-27:
You have X_SLAVE_ADDRESS set to 3. Each stepper has it's on software serial pins on the SKR Pro, so normally you would leave these all with the default `0` value and remove all the jumpers underneath the stepper driver.

boelle, 2019-11-27:
@DJDOUBLEG so issue solved? did you check those links?

boelle, 2019-11-27:
since you gave a thumbs up i will consider it solved and close the issue

DJDOUBLEG, 2019-11-27:
> @DJDOUBLEG so issue solved? did you check those links?

No. :( 
I tried older version and work. With new version Marlin (latest) not work. (no HW problem)

DJDOUBLEG, 2019-11-27:
Noooooo

boelle, 2019-11-27:
then why did you do thumbs up?

DJDOUBLEG, 2019-11-27:
Sorry. FB attitude...

boelle, 2019-11-27:
so what version do you use? give precise link to where you download marlin from

did you look at the links me and @shitcreek gave?

DJDOUBLEG, 2019-11-27:
https://drive.google.com/file/d/1iIGTKDvUlDGvfDGDyCJs-XPgFCwB1c0k/view?usp=sharing

boelle, 2019-11-27:
that is not where you should download marlin from

did you look at the links me and @shitcreek gave?

DJDOUBLEG, 2019-11-27:
Sorry. I make post in Bigtreetech FB group. Then Ian Leslie send me our working Marlin in zip.
Not i downloaded from any source.

boelle, 2019-11-27:
we cant help you if you are using outside copy of marlin, so please ask help in Bigtreetech FB group

DJDOUBLEG, 2019-11-27:
:(

boelle, 2019-11-27:
its only fair that we only support marlin and not copies/forks

its like driving a BMW and expect Mercedes to do support work

DJDOUBLEG, 2019-11-27:
[image]

boelle, 2019-11-27:
yes but you got marlin from an outside source on facebook

boelle, 2019-11-27:
your copy can have many changes etc we dont know about

DJDOUBLEG, 2019-11-27:
I understand, I just wanted to indicate that it is not a HW bug because there is a version that worked.
We should not focus on what works but what does not.

boelle, 2019-11-27:
you need to ask for help where you got marlin from (FB)

DJDOUBLEG, 2019-11-27:
Please forget that I sent a link to a working version.
Let's focus on why the last version doesn't work.

boelle, 2019-11-27:
sorry, go to FB and ask for help

DJDOUBLEG, 2019-11-27:
LOL

DJDOUBLEG, 2019-11-27:
WHY?

boelle, 2019-11-27:
because that is where you got marlin from

DJDOUBLEG, 2019-11-27:
The latest version i have rproblem, This version from Marlin official Github repo!!!

DJDOUBLEG, 2019-11-27:
I download from this link: https://github.com/MarlinFirmware/Marlin/archive/1.1.x.zip

boelle, 2019-11-27:
use 2.0.x it will most likely fix the problem

boelle, 2019-11-27:
https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x

DJDOUBLEG, 2019-11-27:
Sorry: From this link:https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip
About 1 hour.

DJDOUBLEG, 2019-11-27:
I tried from FB source only test. Not clear for me is HW or FW problem.  From FB source have work but very old source. And official source new but not work.

sjasonsmith, 2019-11-27:
@DJDOUBLEG, I already told you exactly what your problem probably is. Did you even try it?

DJDOUBLEG, 2019-11-27:
Yes. I tried this. But not work. :(
Same issue.

Jason Smith <notifications@github.com> ezt írta (időpont: 2019. nov. 27.,
Sze 21:11):

DJDOUBLEG, 2019-11-27:
It was unnecessary to open this ticket.

closed 2019-11-27 by DJDOUBLEG