MarlinFirmware/Marlin #16042: [BUG] Autohome not working if "Z_SAFE_HOMING" is deactivated

uttrichp, 2019-11-28:
After commenting out "#define Z_SAFE_HOMING" auto homing does not work anymore. The homing of  X and Y is done correctly, both switches are triggered. Some seconds after Y homing is finished, BL Touch is will be deployed and the z homing starts without moving X back to the print bed. So basically the nozzle is places left of the print bed and will never trigger. The probe is offset is 0,-38,0. 

As soon as I am activating "#define Z_SAFE_HOMING", everything is working as expected. 

I am Using a Bigtreetech SKR V1.3 with the actual git. 
 
Expected behavior: Probe/Nozzle is moving away from the X min switch to the center and starts Z homing-

Actual behavior: Probe/Nozzle stays at the position where X min switch has been triggered and starts Z homing left of the print bed.

[config.zip](https://github.com/MarlinFirmware/Marlin/files/3903237/config.zip)

AnHardt, 2019-11-29:
That's exactly what `Z_SAFE_HOMING` is for - homing Z in the middle of the bed, instead of where it is left after homing the other axes.

Why don't you want to set `Z_SAFE_HOMING`?

uttrichp, 2019-11-29:
I am very sorry, we can close this one - My Bad

I am needing negative Z-axis offsets and mixed it with #define MIN_SOFTWARE_ENDSTOPS. 

And no, I am not new to Marlin ;) 
I am hunting on this for about 12h - This was a combination of arrogance and hysterical blindness.

closed 2019-11-29 by uttrichp