MarlinFirmware/Marlin #16043: [BUG] Baby step double click not working

Neiljclements, 2019-11-29:
I am using an SKR1.3 with a 12864 Mini RGB display and it won’t allow you to access baby steps via a double click of the encoder.

When you try and double click it just returns to the info screen. I have tried clicking multiple times and at varying speeds, but nothing works.

I have read of similar issues with the 12864 Mini (non RBG) where is was solved by increasing the buffer size to 64 or 128, which I have tried, but it still won’t work.

However, I think the issue is similar to a previous bug report where the menu options couldn’t be accessed when the printer began printing and increasing the buffer size improved it. It doesn’t appear to be the case with the RGB version of the 12864 Mini.

I can access the Z offset menu via the ‘Tuning’ menu option once the print has begun. 

### Description

### Steps to Reproduce
**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files. 

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Vertabreak, 2019-11-29:
to get the double click menu to work on my board i had to up the delay time for like 2 seconds for it to work hope this helps.

Neiljclements, 2019-11-29:
Was this with the Mini RGB? Also, whilst I really appreciate the suggestion, which I will definitely be trying, that’s more of a workaround than a fix. I’m hoping there’s a proper fix so it operates like other displays. Thank you for the speedy reply. I’ll let you know if that works for me

Vertabreak, 2019-11-29:
until i change the delay to be longer i simple was unable to click it fast enough to trigger it seems more like a over sight then a bug.

Neiljclements, 2019-11-29:
Ahhhhh, so you’re saying it’s likely to do with double click timing, which you solved by increasing the delay to 2 seconds. Sorry, I misunderstood what you meant. So, you set it to 2000?

Vertabreak, 2019-11-29:
i did but you may wish to try 3k or 4k if its still to short.

boelle, 2019-11-29:
this is most likely to be an issue of not doing the double click correct

you just have to get the click speed right

will close and we can reopen if it turns out to be a bug as such

closed 2019-11-29 by boelle

Neiljclements, 2019-11-30:
Yes, I'm doing the double click correct.  I have four other printers I use baby steps regularly and I have tried numerous times, at numerous speeds.
I have tried chaging the double click interval all the way up to 5000ms and it still won't work.

The only way I can access it is via the Tuning menu option.