MarlinFirmware/Marlin #16046: MKS Robin tft V2.0 don't work-  There are no buttons to move in the menus.

NachoFrutos, 2019-11-29:
Hello, today I have downloaded the latest version of marlin 2.0. I have compiled and generated the .bin file for the mks_robin board. And on the TFT screen you only see the information of the values. but noy there are buttons to enter the menus. attached configuration files.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3906112/Marlin.zip)

shitcreek, 2019-11-29:
The Marlin status screen is meant for displays with a dial button - it has no on screen button - even if it does, it would not be touchable.  You'll need to use the touch interface that came with your TFT. Here is a great how-to tutorial by Chris' Basement: https://www.youtube.com/watch?v=f651bIg4SCo

thisiskeithb, 2019-11-29:
@shitcreek These boards have a tethered "dumb" tft that when flashed with stock Marlin, they revert to the basic UI with touch buttons underneath (something like this, it's probably changed):

[image]

They worked when I originally submitted the pins for the Robin boards, but code around the TFT has changed enough to break and I don't have physical hardware to test with.

shitcreek, 2019-11-29:
I stand corrected.

thisiskeithb, 2019-11-29:
It's also what is holding up working Tevo Nereus configs unless @zillarob has some good news.

KosVas91, 2019-11-30:
`//#define TOUCH_BUTTONS`
According to your config, you forgot to enable Touch Buttons.

boelle, 2019-11-30:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

KosVas91, 2019-11-30:
> 
> 
> Configurations, please
> Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
> requested in the Issue template) and drop them into your next reply.
> We'll check them over and see if anything is amiss.

Topic starter attached config to the first post... and TOUCH_BUTTONS are not defined there.

boelle, 2019-11-30:
uppsi, did not see the link to config :-(

zillarob, 2019-12-02:
> It's also what is holding up working Tevo Nereus configs unless @zillarob has some good news.

I havent played with that one for a bit, may have to break it out again.

Like Kos said above though, he needs to start by enabling the buttons.

tpruvot, 2019-12-02:
there is a sample config in marlin

boelle, 2019-12-02:
We need actual config and not example ones??? Or did you mean the example config was a starting point?

tpruvot, 2019-12-02:
i tested it ok during the last fix... with the sample in marlin tree, the one also used for travis test..

boelle, 2019-12-02:
so is this one an user error with the config?

tpruvot, 2019-12-02:
https://drive.google.com/file/d/19zjFjk2XoG93TN7go9lv5q_XIunrDD6u/view?usp=sharing

boelle, 2019-12-02:
why a link to picture? you can just paste it here

boelle, 2019-12-02:
[image]

tpruvot, 2019-12-02:
tx, i cant copy from google drive... unsure why

boelle, 2019-12-02:
open in windows paint, ctrl+a followed by ctrl+c
then here ctrl+v

boelle, 2019-12-02:
but my question remains, is there a bug in marlin or is it a user error (config error)

tpruvot, 2019-12-02:
look like an user side error, if he didnt enable touch buttons... cant be working touch buttons

boelle, 2019-12-02:
oki will close then

we can reopen if its a true bug

closed 2019-12-02 by boelle

NachoFrutos, 2019-12-02:
Hello, activating the #define TOUCH_BUTTONS option. the buttons appear but now it does not work when the button is pressed. I guess you have to calibrate the screen because with the values ​​of mks robin tft v2.0 it doesn't work. Is there any documentation or video that explains how to calibrate the screen? thanks

tpruvot, 2019-12-02:
even with wrong calibration values, the screen should be clickable somewhere... mirrored or rotated...

I did my test with the pasted values, and as you can see on the picture, its the V2.0 too

thisiskeithb, 2019-12-03:
Makerbase updated their [Robin repo](https://github.com/makerbase-mks/MKS-Robin) recently, so could someone check if their version works still? I don't own any Robin/Robin Nano boards, so I can't help debug on that front.

If that code works, then it should be easy to port whatever settings/changes they made to get these tethered TFTs working properly on all Robin/Robin Nano boards.

NachoFrutos, 2019-12-03:
Hi, I deleted all the files and downloaded the latest version marlin again. I have compiled with the sample files of mks robin. And everything works. Thank you very much. It can be completely closed. The problem has been user configuration error. I am very sorry for the waste of time and thank you all very much.