MarlinFirmware/Marlin #16051: Marlin 2.0.xbugfix downloaded 11-29-2019

brosskgm, 2019-11-30:
During compile for Anycubic Kossel Linear Plus it produces two errors. With no explanation how to fix can be found.

Marlin/src/HAL/HAL_LPC1768/../../sd/../inc/SanityCheck.h:251:4: error: #error "Remove DELTA_CALIBRATION_RADIUS and use MIN_PROBE_EDGE to inset the probe area instead."
#error "Remove DELTA_CALIBRATION_RADIUS and use MIN_PROBE_EDGE to inset the probe area instead."
^~~~~
Marlin/src/HAL/HAL_LPC1768/../../sd/../inc/SanityCheck.h:1064:4: error: #error "CLASSIC_JERK is required for DELTA and SCARA."
#error "CLASSIC_JERK is required for DELTA and SCARA."

Thanks

sjasonsmith, 2019-11-30:
Are you using old config files? I suggest you compare your files with the current examples to see what has changed.

In most cases, you just need to delete your DELTA_CALIBRATION_RADIUS. MIN_PROBE_EDGE is probably already set to something reasonable, like 10mm.

For `CLASSIC_JERK`, just search for that in the configuration files and enable it.

brosskgm, 2019-11-30:
I actually used the configs in the examples for kossel and was working on the needed changes for the SKR v1.3 board. When I used 2.0 for TriGorilla it worked/Compiled great couple weeks ago. I originally had 98 errors until I updated the platformio.ini with the one updated 8 days ago and it leaves these two left.

sjasonsmith, 2019-11-30:
Those examples have changed in the past couple weeks. The Anycubic Kossel example builds cleanly for me, as it was updated to work with the latest code changes.

Make sure you are updating all of your files when updating to a newer state of bugfix-2.0.x. That includes examples, config files, etc.

brosskgm, 2019-11-30:
Yep. Just downloaded it again and starting over from scratch for a second time.

Thank for your reply.

brosskgm, 2019-11-30:
Well, improvement. Only one error and I have no idea on this.

{
	"resource": "/Marlin-bugfix-2.0.x/.pio/libdeps/LPC1768/U8glib-HAL_ID1932/src/clib/u8g_com_i2c.c",
	"owner": "cpp",
	"severity": 4,
	"message": "'u8g_i2c_opt' defined but not used [-Wunused-variable]",
	"startLineNumber": 44,
	"startColumn": 16,
	"endLineNumber": 44,
	"endColumn": 16
}

Vertabreak, 2019-11-30:
the error suggest your missing library u8glib.

ManuelMcLure, 2019-11-30:
That’s just a warning- it shouldn’t cause the build to fail.

Vertabreak, 2019-11-30:
if it builds it fine in general.

ManuelMcLure, 2019-11-30:
That warning can be completely ignored. It’s just an unused variable definition.

ebraiman, 2019-11-30:
I'm also seeing compile issue from a fresh download today.  
Steps:
Download a fresh marlin firmware version 2.0.x alpha build
Open in ATOM/PlatformIO
configuration.h:
change   #define MOTHERBOARD BOARD_BIGTREE_SKR_V1_3
change  #define SERIAL_PORT -1
platformio.ini:
change default_envs = LPC1768
expected result:
compiles without issue.
actual result:
compile fails first time.
compile passed second or third time without files modified further.

brosskgm, 2019-11-30:
Ok Thanks.

After compiling it a second time that warning went away. Now to setup everything else.

Thanks
Bob

boelle, 2019-11-30:
@brosskgm so this one is solved allready?

brosskgm, 2019-11-30:
Yes it is. Just had to start over.

closed 2019-11-30 by brosskgm