MarlinFirmware/Marlin #16052: [BUG] TB6600 linear shift layer on X and Y axis

F4HTB, 2019-11-30:
### Description

We built a homemade printer.
We use an A8 motherboard on which we take the command: motor driver enable, step and direction.
We use a stage of amplification to pass the signals of the card anet A8 towards the drivers.

We use cura 4.4 or 4.3 to slicer.

**When we use marlin 1.1.8, the prints are good.**

**When we use marlin 1.1.9 or 2.0, we have the same problem.**
When a layer is made, when returning to the original position for the next layer, it appears that one or more steps of the thickness of a layer are not performed.
This does not translate into a regularized layer shift.
See picture.

We reviewed most of the parameters around drivers and possible EMI problems etc ...

It seems that since version 1.1.9, a bug is introduced in the code.

On the reprap.org forum, I read several posts about this problem and it be never solved by another solution than to return to version 1.1.8

Attach the parameter files is not needed, we have try all parameters, we get We usually have the same result or worse. 
But quite honestly, someone really needs to look at the TB6600 drivers to see why it works in 1.1.8 and not in 1.1.9, 1.1.X-bugfix or 2.0 ....

[image]
[image]

Big Thanks in advance.

thisiskeithb, 2019-11-30:
Did you initialize your `EEPROM` after updating? Run `M502`, followed by `M500` to perform a [factory reset](http://marlinfw.org/docs/gcode/M502.html).

F4HTB, 2019-11-30:
I am not but hard reset after flashing made this no?

sjasonsmith, 2019-11-30:
This is probably related to other issues with missed steps on DRV8825 drivers. We're probably generating step pulses smaller than the minimum the TB6600 needs.

The TB6600 requires a 2.5 microsecond step pulse, and I suspect we are generating pulses that are sometimes only about 1.4 microseconds.

I suggest setting `MINIMUM_STEPPER_PULSE` in Configuration_adv.h to 4 or even 5, to see if that helps.

thisiskeithb, 2019-11-30:
Jason is probably closer to the real answer with his reply, but running `M502` & `M500` after flashing firmware is good practice. Especially when moving through firmware versions while testing.

F4HTB, 2019-11-30:
Already tested with 2, 3, 4, 5 10, 30.
No change...

F4HTB, 2019-11-30:
Thank thisiskeithb, I would not fail to keep this in mind for nexts tests.

F4HTB, 2019-11-30:
Same problem here for example:
https://reprap.org/forum/read.php?415,838766

boelle, 2019-11-30:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

F4HTB, 2019-11-30:
Hi boelle, 

Thanks for reply.

See attached differents tests in 1.1.x (1.1.9 bug fix).
Tell me if you want 2.0 versions but the changes are almost the same.

[config tests.zip](https://github.com/MarlinFirmware/Marlin/files/3907213/config.tests.zip)

F4HTB, 2019-11-30:
See attached last test in 2.0
[Last tests in Marlin 2.0 TB6600.zip](https://github.com/MarlinFirmware/Marlin/files/3907214/Last.tests.in.Marlin.2.0.TB6600.zip)

F4HTB, 2019-11-30:
After some other research, i have unmount my driver and it seem to be some TB67S109AFTG.

https://datasheet.lcsc.com/szlcsc/1811151451_TOSHIBA-TB67S109AFTG_C92125.pdf

(TB6600 https://datasheet.lcsc.com/szlcsc/1809050226_TOSHIBA-TB6600HG_C66042.pdf)

But from what I read they go up to 200khz so that does not seem to be worse than a TB6600 ...
And that does not solve the problem of why in 1.1.8 it works but not in 1.1.9 and 2.0 ...

boelle, 2019-11-30:
> Tell me if you want 2.0 versions

1.1.9 and older are not updated anymore so you should always use 2.0.x

> why in 1.1.8 it works but not in 1.1.9 and 2.0

maybe a mistake was made, and toshiba drivers are not used much so it takes a long time before mistakes and errors are discovered

cjsoong, 2019-12-03:
@F4HTB 
I use TB6600 in XYZ and it works fine.
This is my config for your reference
board type: MKS SGEN_L
[TB6600_config.zip](https://github.com/MarlinFirmware/Marlin/files/3915431/TB6600_config.zip)

boelle, 2019-12-04:
@F4HTB have you tried the config from @cjsoong ?

also remember marlin 2.0 was released yesterday

F4HTB, 2019-12-06:
@boelle , yes i do. 

I had to adapt some parameters like those of the end of races etc ... But always the same problem.
Also, our machine is a prototype and I do not always have it at hand.
To be certain of the problem, we will take a digital oscoloscope and record all the commands signals on one of the axes that does not work.
It will take time but at least we will know if the minimal time instruction of the steps is respected or not.
Also and at the same time, I will try to modify the controllers. They have 4n25 inputs and it is possible that they are limiting. So we will change that with a more efficient optocoupler.
I will share my advancements as soon as possible.
Thank you for leaving this post open until I have something new.

boelle, 2019-12-06:
i know that @sjasonsmith is doing some experiments and have put in this PR # 16128

boelle, 2019-12-06:
and yes version 2.0.1  will be released some time this weekend, so i think we can expect far more updates and the issue list is not down to ~85 open issues

sjasonsmith, 2019-12-06:
> i know that @sjasonsmith is doing some experiments and have put in this PR #16128

I don't think my changes will solve this problem, since @F4HTB mentioned that changing MINIMUM_STEP_PULSE did not help.

boelle, 2019-12-24:
btw, who can confirm there is an issue? use the same configs as OP and the same hardware if possible

F4HTB, 2019-12-24:
Hello dear boelle, 
I go to record signals the current the next week.
At the same time i will try the new version.
happy holidays and see you next week.

boelle, 2020-01-07:
@F4HTB what did you record, its been slightly over a week

Takker3D, 2020-01-13:
Hi, gays! We have this problem too. Board Arduino DUE + RuRamps4D v1.3 + 2 drivers 2HSS57 (2 pcs on XY axes), H-bot kinematics. Firmware is "Marlin 2.0.x bugfix". 
  We tried to change "MINIMUM_STEP_PULSE" too, but it didn't work. 
  Thank You for your help.

boelle, 2020-01-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

F4HTB, 2020-02-04:
Hi, sorry for the time i took for the first messures.

Attached is a zip file containing the marlin configuration files 1.1.8 which works and 2.0.X where there is the offset.

I also put the frames recorded on the X axis by a logic analyzer.
You must use the Saleae 1.2.18 software which is free on the internet.
https://www.saleae.com/fr/downloads/

As a reminder, the offset in 2.0.X seems to be the same as in Y. This is why I have only recorded the X axis for now.

We printed the same 10mm cube test in both cases.

I have already noticed that the frames are not necessarily the same and it seems to be missing steps with Marlin 2.0.X

Next week I will record the complete cube on X and Y.

Hope this can help you.
[capture_logique.zip](https://github.com/MarlinFirmware/Marlin/files/4153415/capture_logique.zip)

thinkyhead, 2020-02-04:
The TB6600 is not well known or well supported. Have you tried playing with the stepper timing options to see if it helps?

sjasonsmith, 2020-02-04:
Is the problem on external drivers, not drivers directly on a board? If so, those often use optocouplers with their own timing requirements, which can be much slower that the driver chip itself.

One experiment could be to disable multistepping by adding the following line to the configuration:
`#define DISABLE_MULTI_STEPPING`

F4HTB, 2020-02-04:
@thinkyhead yes i have already try

@sjasonsmith i will try it next week.

sjasonsmith, 2020-02-05:
@F4HTB, can you provide a little more information about this part of your configuration?

> We use a stage of amplification

I already mentioned optocouplers possibly impacting timing requirements, but this amplification stage may have separate timing requirements as well.

I believe I have some TB6600 drivers, but I have only used them with grbl. I could possibly set them up for some testing, but I don't know when I would find time to do that.

Wonderr, 2020-03-27:
Same layer sifting X and Y axis on me.

Wonderr, 2020-03-28:
[https://www.youtube.com/watch?v=0Zx121bECvw
](url)

galonsog, 2020-04-05:
Hi.

I fear the same problem.

Print crooked with the TB6600 drivers.

I have configured the parameters:
Minimum Stepper Pulse 5
Maximum stepper rate 150000
Direction Stepper Delay 1500

There is no change when modifying these parameters.

I have tried to load the firmware from the arduino ide and from VSCode, with the same result.

With other firmware (MK4Duo) it works fine.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4434395/Configuration.zip)

closed 2020-06-20 by boelle