MarlinFirmware/Marlin #16053: [FR] Heatbed maintenance before printing

yudi-jpn, 2019-11-30:
### Description

Heatbed maintenance before printing.
Added bed maintenance menu to menu

### Steps to Reproduce

1. When the bed maintenance menu is selected, the bed is ejected and heated for a certain period of time for cleaning.

2. When a “confirm” operation is performed on the screen menu, auto home is executed and the printer enters a print waiting state.

#### Additional Information

While the bed is hot, use alcohol to remove old resin and oil and dust.
This operation makes it difficult for the object to peel off from the heat bed.

Bergerac56, 2019-11-30:
Why not just create a custom command to do that? So you can do exactly what you need.

Bergerac56, 2019-11-30:
Here a potential custom command (to put in configuration_adv.h). I did not test this command. There could be some mistakes ;): 
 
#define USER_DESC_1 "Bed Maintenance"
  #define USER_GCODE_1 "M190 S" STRINGIFY(PREHEAT_1_TEMP_BED)"\nG28\nG1 Z50 F5000\nG1 X0 Y0 F5000\nG4 S120\nM104 S0\nG28\nM84"

You have to adapt G1 X0 Y0 to reach the position you want and G4 S120 ( wait here 120 seconds) to the desired waiting time

The command: 

- preheat the bed at your default PLA bed temperature 
- Home the printhead 
- Raise the head for 50 mm 
- move the bed to (0,0) position (to change to where you want)
- wait 120 sec (adapt if needed)
- cooldown the bed 
- home the printhead again 
- stop stepper motors

boelle, 2019-11-30:
why not clean the bed cold, a hot bed makes alcohol evaporate faster

Bergerac56, 2019-11-30:
@boelle I agree. This is what I do ;)
But this is what was asked :)

yudi-jpn, 2019-11-30:
@Bergerac56
@boelle
Thank you.

I didn't know about the existence of custom commands.

Even a surfactant could remove some oil film, but it was very convenient to dissolve the old resin with high-concentration alcohol while the heatbed was warmed.

Since ethanol is used, there are few adverse effects other than resin.

Thank you carefully.
Should i close the thread?

Thank you in advance.

AnHardt, 2019-11-30:
The problem with this kind of macros is that they leave a permanent mark. Here the position and  F is changed and there is no way back to the original values. For using this macro before the actual print starts this is ok, bur for using macros during a print a more advanced macro language would be very useful. Maybe in Marlin V3.0.
For now we could try if we can selectively push/pop at least X, Y, Z, E, F, Tool, Retraction_amount Temperature, ...,  and Fan% per G-code. Maybe in more than one level.

yudi-jpn, 2019-11-30:
@AnHardt
Thank you very much.

Although there is a timing relationship, depending on the location and conditions of use, oil and soot in the atmosphere will adhere to the bed.

Some printers vaporize slightly during printing in non-grounded environments, so odors in the air can also solidify and adhere to the bed.

I think menus are maintenance that can be performed easily and regularly before printing.

What I feel may be noisy.

However, I felt a little inconvenient for items that were likely to have the required users.

Thank you carefully.

boelle, 2019-11-30:
> Should i close the thread?

nope, but i think this feature request will be low on list

yudi-jpn, 2019-11-30:
@boelle
I understood. Thank you.

I Check the contents of configuration_adv.h, there may be no personal time.

If I can interpret the structure, I can add it.
(Sad that it is likely to disappear by version upgrade)

I also personally think about how to reduce the warpage of ABS and PETG materials.

Thank you very much for your kindness.

Vertabreak, 2019-12-19:
heating the bed for cleaning can damage some beds for instance there is a coating on my bed and i cleaned it hot once and it dissolved the coating off the bed its just bare glass in some spots now.

closed 2020-06-30 by github-actions[bot]