MarlinFirmware/Marlin #16056: [BUG] Fysetc S6 PWM duty cycle too low

Drachenkaetzchen, 2019-11-30:
### Bug Description

When setting Fan1 speed to 100%, the PWM duty cycle is only 6%.

### My Configurations

[Configuration.h](https://github.com/Drachenkaetzchen/Marlin/blob/PurpleCube-S6/Marlin/Configuration.h)
[Configuration_adv.h](https://github.com/Drachenkaetzchen/Marlin/blob/PurpleCube-S6/Marlin/Configuration_adv.h)

### Steps to Reproduce

1. Set Fan1 to 100%

**Expected behavior:**

PWM duty cycle of 100%

**Actual behavior:** [What actually happens]

PWM duty cycle of 6%

[image]

Drachenkaetzchen, 2019-11-30:
When using `FAN_SOFT_PWM` the duty cycle is 100%, as expected.

sjasonsmith, 2019-11-30:
This is similar to the SKR Pro behavior before the changes @LinoBarreca made, including the 20 kHz frequency you see here.

LinoBarreca, 2019-12-01:
what kind of board is it? which architecture?

LinoBarreca, 2019-12-01:
STM32... that's why. with 5.8.0 they should have fixed it. there was a bug in the duty cycle calculation which I think I saw being handled.

LinoBarreca, 2019-12-01:
meanwhile you can just lower the PWM frequency, as I did in the skr-pro, to 1khz to have the correct duty cycle

Drachenkaetzchen, 2019-12-01:
I think I tried that, but it had no effect. Maybe I messed something up, will try again later to verify

Drachenkaetzchen, 2019-12-01:
I have found the relevant upstream issue:

https://github.com/stm32duino/Arduino_Core_STM32/issues/790

So it's related to 8 bit to 12 bit mapping and not the frequency.

boelle, 2019-12-11:
i just made an PR to fix scara printers and that made me think on other issues

so if you know how to fix this one just let me know what file to change or add (or remove) and i could submit the PR

sorry if this is bad formulated (not the sharpest pencil arround)

Drachenkaetzchen, 2019-12-11:
> i just made an PR to fix scara printers and that made me think on other issues
> 
> so if you know how to fix this one just let me know what file to change or add (or remove) and i could submit the PR
> 
> sorry if this is bad formulated (not the sharpest pencil arround)

it's an upstream issue, see link above

boelle, 2019-12-11:
can marlin as project do anything about that?

Drachenkaetzchen, 2019-12-11:
> can marlin as project do anything about that?

yes, wait for a newer upstream release and update library refs then

Drachenkaetzchen, 2019-12-11:
also, of course, test with the new upstream release and verify that it works correctly. If not, notify the upstream authors and see what we can do to make it work.

dinamitemic, 2020-02-24:
Hi, I've (hopefully) found a solution for this problem. 
As reported in this [PR](https://github.com/stm32duino/Arduino_Core_STM32/pull/814) from the stm32 core there is a problem in the definition of the pwm resolution.
In particular we've to change it in **buildroot\share\PlatformIO\variants\FYSETC_S6\variant.h** 
from:
`#define PWM_RESOLUTION          8`
to: 
`#define PWM_RESOLUTION          12`
Tryed it and now it works as expected. Tested with oscilloscope. [DEMO VIDEO](https://youtu.be/cgMZDhy4dXU)
 I think that from latest updates the default resolution is 12 bits and there is no need to declare it anymore, but not tested yet.

sjasonsmith, 2020-06-24:
This is most likely already fixed. I will test it on my board to be sure.

sjasonsmith, 2020-06-24:
I just tested this and verified that the duty cycle is now correct. Closing this as fixed.

closed 2020-06-24 by sjasonsmith