MarlinFirmware/Marlin #16059: [BUG] mini12864 with Fysetc S6 shifts display contents up after about 30s

Drachenkaetzchen, 2019-11-30:
### Bug Description

After about 30 seconds, the fysetc mini12864 display contents gets shifted up by about half a screen, but continues to function. When returning to the info screen and entering the menu again, the contents are displayed as expected again until 30 seconds later.

The same display works fine on Rumba and SKR 1.3 boards, so I assume it's an issue specific to the Fysetc S6.

### My Configurations

[Configuration.h](https://github.com/Drachenkaetzchen/Marlin/blob/PurpleCube-S6/Marlin/Configuration.h)
[Configuration_adv.h](https://github.com/Drachenkaetzchen/Marlin/blob/PurpleCube-S6/Marlin/Configuration_adv.h)

### Steps to Reproduce

Use a fysetc S6 connected to a fysetc mini12864 display.

1. Wait about 30 seconds
2. Observe that the display contents have shifted

#### Additional Information
[image]

Drachenkaetzchen, 2019-12-01:
I tested with a Reprapdiscount Smart Controller, it does not show that behavior.

Drachenkaetzchen, 2019-12-01:
I also increased the display timings, no effect:

```
#define BOARD_ST7920_DELAY_1 DELAY_NS(206)
#define BOARD_ST7920_DELAY_2 DELAY_NS(102)
#define BOARD_ST7920_DELAY_3 DELAY_NS(1284)
```

Drachenkaetzchen, 2019-12-06:
I have had a further look at this and the Fysetc mini12864 uses SPI. The datasheet for the ST7567 
 controller specifies a maximum of 20MHz for the clock. Gotta get my scope out and see how fast it actually is.

Drachenkaetzchen, 2019-12-07:
Also would be interesting to try `FORCE_SOFT_SPI` to see if it makes any difference. Not sure if it's a timing issue though. The shifting seems too consistent.

Drachenkaetzchen, 2019-12-07:
More test results:

- Issue occurs with 3 or more TMC2130 drivers plugged in. It doesn't seem to matter if the stepstick position is actually enabled. For example, I have X,Y,Z,E0 enabled as TMC2130 and put TMC2130 drivers in X,Y and E1. Issue occurs.
- Issue doesn't seem to occur with only 2 TMC2130 drivers plugged in

Drachenkaetzchen, 2019-12-08:
- `FORCE_SOFT_SPI` results in a blank display, didn't mess around with that any further
- Disabling the driver types or reverting to `A4988` doesn't do anything
- The Fysetc fork of Marlin shows similar behavior

Right now I can't explain the issue differently than some weird signal reflections.

Drachenkaetzchen, 2019-12-08:
The measured SPI Clock is about 1.4MHz.

Drachenkaetzchen, 2019-12-08:
Looking through the `u8g2` code it seems that the `ST7565_START_LINE` is only used during initialization. So if there's noise or reflections on the line and the display interprets the `ST7565_START_LINE` by that, the display gets shifted but never reverts back to start line 0. To me that's the most plausible explanation right now.

Drachenkaetzchen, 2019-12-08:
Potentially related fixes:

https://github.com/MarlinFirmware/Marlin/commit/23bb40d0ea398cee3f68bd9a4a716a3b7dda8db9
https://github.com/MarlinFirmware/Marlin/issues/13550

Drachenkaetzchen, 2019-12-09:
I have to stop debugging this as the TMC drivers are needed in my other machine. I updated the initial report with my findings

iankds05, 2020-02-13:
For what it is worth one of the issues I have found with RepRap type displays is that they are much more susceptible to jitter and noise in the ribbon cables connected to 32bit MPU boards than they ever were on 8bit ones, especially if run in proximity to any other cables, either in or outside of a harness. On 8bit Atmel 2560 boards I have run the displays over a meter away from the main control board with no issues but with 32bit MPU boards anything over 150mm on the ribbon cables causes display disfunction either in the form of scrambled pixels or jumpy response to menu selections. One successful solution I have found is to abandon the ribbon cables, and customise new cables using 28AWG single cores which still fit into the IDC 10 way connectors and by twisting the 10 cores together and placing a ferrite core on one end nearest the display all issues seem to be eliminated. Weather this would be a possible solution to this particular issue I cannot be sure as I don't posses the afore mentioned display to test it thoroughly

AnHardt, 2020-02-13:
It's not the difference 8 to 32 bit. It's 5 vs. 3.3V, lowering the noise distance and making the ST7920 even slower (see [datasheet](http://www.hpinfotech.ro/ST7920.pdf) Page 43-44).

jonathanlundstrom, 2020-03-25:
Has this problem been fixed in Marlin 2.0.5.2?

jonathanlundstrom, 2020-04-22:
This problem is still highly prominent. I also noticed that I'd you've got the speaker/buzzer enabled in `Configuration.h` then motor movements will not work properly. This will also cause the jitter to increase severely.

While debugging yesterday I also ran into another problem. While printing over SD the printer just decides to skip chunks of code which results in some very strange results. I've tried several models through several SD-cards but the problem only occurs though SD. It works fine over USB.

sjasonsmith, 2020-07-08:
I re-opened this based on comments from @jonathanlundstrom on Discord that this is still an issue.

closed 2020-08-14 by github-actions[bot]