MarlinFirmware/Marlin #16062: LCD RADDS

boubouexpress, 2019-12-01:
hello

during printing I can not use the LCD menu

merci
[marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3908261/marlin.zip)

shitcreek, 2019-12-02:
Have you tried increasing the `BLOCK_BUFFER_SIZE`?

boubouexpress, 2019-12-03:
Yes go to 32

boelle, 2019-12-04:
is this a bug or not????

[image]

boelle, 2019-12-07:
will close this one until er hear from @boubouexpress 

we need to know if this is a bug or not, and what is wrong

boubouexpress, 2019-12-07:
it's a bug
my screen does not react anymore when print and in court

boubouexpress, 2019-12-07:
functions perfectly before printing
after the hot and finish and it starts to evect the printing movement
it correctly display the information but I can not access the menu anymore

boubouexpress, 2019-12-07:
sorry for the translation

boubouexpress, 2019-12-07:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3935065/Configuration.zip)

boelle, 2019-12-07:
oki, that makes a bit more sense

and i say sorry for not understanding better :-)

boubouexpress, 2019-12-07:
it's normal with google translation
thank you

AnHardt, 2019-12-07:
I can hear your processor screaming, begging for mercy.

#define XYZ_MICROSTEPS 128
#define MIN_STEPS_PER_SEGMENT 1
#define BLOCK_BUFFER_SIZE 16

boubouexpress, 2019-12-07:
hello
WHAT VALUE I MUST MASTER

thank you

AnHardt, 2019-12-07:
It's not one of them. It's the combination.
All of them, plus '#define DELTA_SEGMENTS_PER_SECOND 200' contribute.

\#define XYZ_MICROSTEPS 128 keps the processor busy with stepping - having less time for the planner.
\#define MIN_STEPS_PER_SEGMENT 1 allows very short segments and many of them. Each needing its time to plan.
\#define DELTA_SEGMENTS_PER_SECOND 200 allows more and shorter segments than usual. (each only lasting 1/200 second)
\#define BLOCK_BUFFER_SIZE 16 is very short for the above combination. The display updates only when there is enough time in the buffer, so buffer is not running dry during the update..

boubouexpress, 2019-12-07:
I modify
#define XYZ_MICROSTEPS 32
#define DELTA_SEGMENTS_PER_SECOND 160
the rest I do not change, I'm not really what value meter
if you have to increase or decrease
#define MIN_STEPS_PER_SEGMENT 1
#define BLOCK_BUFFER_SIZE 16

still more to do a try

boubouexpress, 2019-12-07:
hello
a big thank you it works perfectly

closed 2019-12-07 by boubouexpress