MarlinFirmware/Marlin #16063: [BUG] Fysetc S6 Overview Ticket

Drachenkaetzchen, 2019-12-01:
This ticket tracks open issues for the Fysetc S6 boards. Check the individual issues for possible workarounds.

- [ ] [Servo Support](https://github.com/MarlinFirmware/Marlin/issues/16055)
- [ ] [PWM Duty Cycle (affects fans)](https://github.com/MarlinFirmware/Marlin/issues/16056)
- [ ] [M500 EEPROM Emulation not working](https://github.com/MarlinFirmware/Marlin/issues/16060)
- [ ] [mini12864 display content shift](https://github.com/MarlinFirmware/Marlin/issues/16059)

boelle, 2019-12-07:
i see you have created more issues since this one, rememeber to add them to the list :-D

Drachenkaetzchen, 2019-12-07:
> i see you have created more issues since this one, rememeber to add them to the list :-D

If they are related to the S6 only, I will. However, that's not the case.

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-20 by boelle