MarlinFirmware/Marlin #16070: 2.0.x Compile error

effernity, 2019-12-02:
# Bug Description
Marlin does not compile for SKR 1.3

Error:
`In file included from Marlin\src\module\../inc/MarlinConfig.h:45,
                 from Marlin\src\module\configuration_store.h:24,
                 from Marlin\src\module\configuration_store.cpp:47:
Marlin\src\module\configuration_store.cpp: In static member function 'static void MarlinSettings::report(bool)':
c:\marlin\marlin\src\core\serial.h:181:65: error: too few arguments to function 'void serialprintPGM(const char*)'
 #define _SELP_3_P(a,b,c)     do{ _SEP_2_P(a,b); serialprintPGM(c); }while(0)
                                                                 ^
c:\marlin\marlin\src\core\serial.h:183:49: note: in expansion of macro '_SELP_3_P'
 #define _SELP_5_P(a,b,V...)  do{ _SEP_2_P(a,b); _SELP_3_P(V); }while(0)
                                                 ^~~~~~~~~
c:\marlin\marlin\src\core\serial.h:177:30: note: in expansion of macro '_SELP_5_P'
 #define __SELP_N_P(N,V...)   _SELP_##N##_P(V)
                              ^~~~~~
c:\marlin\marlin\src\core\serial.h:178:30: note: in expansion of macro '__SELP_N_P'
 #define _SELP_N_P(N,V...)    __SELP_N_P(N,V)
                              ^~~~~~~~~~
c:\marlin\marlin\src\core\serial.h:204:35: note: in expansion of macro '_SELP_N_P'
 #define SERIAL_ECHOLNPAIR_P(V...) _SELP_N_P(NUM_ARGS(V),V)
                                   ^~~~~~~~~
Marlin\src\module\configuration_store.cpp:3127:9: note: in expansion of macro 'SERIAL_ECHOLNPAIR_P'
         SERIAL_ECHOLNPAIR_P(
         ^~~~~~~~~~~~~~~~~~~
c:\marlin\marlin\src\core\serial.h:271:6: note: declared here
 void serialprintPGM(PGM_P str);
      ^~~~~~~~~~~~~~
Compiling .pioenvs\LPC1768\src\src\module\planner.cpp.o
*** [.pioenvs\LPC1768\src\src\module\configuration_store.cpp.o] Error 1`

### My Configurations

[config.zip](https://github.com/MarlinFirmware/Marlin/files/3908908/config.zip)

### Steps to Reproduce

1. Download last commit (2.0.x release)
2. Compile with my configs

**Actual behavior:** log above

sl1pkn07, 2019-12-02:
https://github.com/MarlinFirmware/Marlin/commit/9b6d2da9bf93280c81170bc181e1b539d29561ed

ellensp, 2019-12-02:
Confirmed on todays marlin bugfix 2.0.x  Enabling ?_DRIVER_TYPE  TMC2208 on any axis is triggering this error.
Reversed    9b6d2da, still errors.

p3p, 2019-12-02:
appears to be an issue with the `SERIAL_ECHOLNPAIR_P` macro `SERIAL_ECHOLNPAIR` works fine when used in that part of `configuration_store.cpp`

thinkyhead, 2019-12-02:
Patched. While C/C++ code is forgiving of extra commas, the preprocessor is less so.

closed 2019-12-02 by thinkyhead

feifei00, 2019-12-12:
I have the same erro report, for i set X aix driver A4988, Y: TMC2208, Z:  A4988, E0: A4988. then i changed all to A4988 or TMC2208, everything is ok.

thisiskeithb, 2019-12-12:
> I have the same erro report, for i set X aix driver A4988, Y: TMC2208, Z: A4988, E0: A4988. then i changed all to A4988 or TMC2208, everything is ok.

Pull the latest [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) code, Clean your PIO build, then Build.