MarlinFirmware/Marlin #16075: [FR] (Auto cooldown extruder if not used)

italocjs, 2019-12-02:
Hi,  i had noticed a issue in particular on the machines we have, that when people heat the nozzle for whatever reason (like remove or change filament, just extrude, or preheat) and forget it heated the plastic degrades (turns brown or black and carbonize) sometimes clogging the printer.  

Is it possible to create a new config on where we could choose a timeout to cooldown the hotend if the printer doenst perform any extrusion movement?

randellhodges, 2019-12-02:
A heater timeout would be a good feature. 

In the mean time, if you have Octoprint connected, there is a heater timeout plugin available that do this.

italocjs, 2020-06-24:
This feature has been developed and its available on latest bugfix, Closing.

closed 2020-06-24 by italocjs