MarlinFirmware/Marlin #16076: [BUG] Linear Advance K Calibration Print stops extruding - Marlin 1.1.9-bugfix

stantond, 2019-12-02:
### Bug Description

Using Marlin 1.1.9-bugfix on an Ender 3 Pro with Creality 1.1.5 board.
Created GCODE for calibrating Linear Advance 1.5 using http://marlinfw.org/tools/lin_advance/k-factor.html
The first test line prints, then the extruder stops.

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

[Linear Advance 0 to 2 kfactor.txt](https://github.com/MarlinFirmware/Marlin/files/3913149/Linear.Advance.0.to.2.kfactor.txt) _.gcode_

[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/3913169/Configuration.txt) _.h_

[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/3913170/Configuration_adv.txt) _.h_

### Steps to Reproduce

1. Generate GCODE using the online tool
2. Print it
3. Watch the extruder

**Expected behavior:** [What you expect to happen]

The extruder extrudes and the full print prints.

**Actual behavior:** [What actually happens]

The first test line (K=0) prints.
The plastic thins out over the next two lines to nothing, and the extruder stops turning entirely.

* Provide pictures or links to videos that clearly demonstrate the issue.

[image]

Grogyan, 2019-12-02:
Please try Marlin 2.0

shitcreek, 2019-12-02:
Marlin 1.1.9 is no longer actively worked on. Please try 2.0. 

I don't see anything that stands out in your config and gcode files. Has the bed been leveled? How exactly does it stop? Like does it continues to try and complete the job except with zero extrusion or does it ends the job after the 3rd line. Have you try doing 0.1 K factor stepping and see if it will produce more lines? Try enabling LA_DEBUG and check the outputs.

stantond, 2019-12-02:
The bed is pretty level as it is, but there's also a G29 in the print GCODE.

It continues and completes the print job, but it stops extruding and doesn't continue. The photo shows a "finished" print.

I can try 0.1 stepping tomorrow, although I suspect it stops as soon as K >0.

How do I use LA_DEBUG? I usually use with Octoprint.

shitcreek, 2019-12-03:
I haven't used Octoprint so I don't know with that. You need to connect to the machine via USB. You could use repetier, adruino ide or whathaveyou. `LA_DEBUG` must be enabled in your configuration first and your machine reflashed - you might as well upgrade to 2.0 :D

boelle, 2019-12-03:
@stantond did you try 2.0.x ?

Strahler85, 2019-12-03:
Could it be, that the 1.1.5 creality board is the silent board with the tmc 2208 on all axis in standalone and stealthchop mode?
As far as a I know, LA does not work with tmc 2208 standalone at Extruder Motor in stealthchop Mode.

thisiskeithb, 2019-12-03:
> How do I use LA_DEBUG? I usually use with Octoprint.

There's a terminal tab in OctoPrint which is where you'll watch/send commands:
[image]

boelle, 2019-12-04:
closing this one until we hear back from @stantond 

as said above 1.1.9 is no longer active worked on, please use 2.0.x

stantond, 2019-12-04:
> As far as a I know, LA does not work with tmc 2208 standalone at Extruder Motor in stealthchop Mode.

@Strahler85 I'm using the Creality 1.1.5 board, and I believe it does use 2208 drivers in stealthchop mode for all motors. Do you know of there's a way to change the mode for the E driver?

@boelle of the above is true, this issue will affect any version of Marlin using this board

shitcreek, 2019-12-06:
In 2.0, you can disable stealthchop for each stepper in `configuration_adv.h` ie `  #define STEALTHCHOP_E`

boelle, 2019-12-07:
i cant reproduce and i think the difference is that i dont use stealthchop on the extruder

https://github.com/boelle/Marlin/tree/Mendel90

i use drv8825 for the extruder and that is why i know, for everything else i use tmc2100

will close this one as it most likely a config issue and not a bug as such

stantond, 2019-12-07:
Hopefully I'll have time to do some more testing on this next week.

I'd suggest that if it is a configuration issue, it's a "bug" in the documentation - if it's a known and understood issue that can be resolved with a config change, adding it to the docs so that the next poor sod to get stuck can find a fix rather than spend hours messing about and trawling closed issues would be a useful "fix", right?

boelle, 2019-12-07:
that's an arguement i can use, so yes, documentation should be improved if it turns out that you cant use stealthchop on the extruder

Vertabreak, 2019-12-10:
i have tmc2208 in stand alone running marlin 2.0 and linear advance function perfectly i ran the calibration 10x yesterday and i use linear advance daily.

stantond, 2019-12-12:
I've just installed Marlin 2.0, with almost identical settings.

M569 isn't working (Unknown command: "M569"), so I've obviously missed something. I've subsequently installed the TMCStepper library as specified [here](http://marlinfw.org/docs/hardware/tmc_drivers.html), but I've been staring at the page for a good few minutes and can't figure out what setting I need to include the library when compiling and get M569 working. Which settings do I need?

thisiskeithb, 2019-12-13:
> I've just installed Marlin 2.0, with almost identical settings.

Can you post updated configs? Just zip them up into a single .zip and attach it and we can verify your settings. I'm guessing you didn't enable `TMC2130` for the driver type, so `M569` isn't enabled.

Also, PlatformIO should download TMC libraries for you without having to install anything if you're using VSCode.

stantond, 2019-12-13:
I tried defining some Stepper Drivers as `TMC2208` and got compiler errors (using Arduino IDE, no idea what PlatformIO is).

Files attached.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3960451/Marlin.zip)

boelle, 2019-12-13:
> no idea what PlatformIO is

try it

stantond, 2019-12-13:
> > no idea what PlatformIO is
> 
> try it

I'm sure it's great, but I'd rather fix what I can without adding more variables to the mix. Maybe when this issue is resolved.

> I tried defining some Stepper Drivers as `TMC2208` and got compiler errors

Changed to `TMC2208_STANDALONE` and it compiles OK. I get the feeling that the Creality board is fixed in standalone mode for all drivers, as they would have just dropped in the new drivers to replace the old rather than design and manufacture a new PCB. Will test this over the weekend.

thisiskeithb, 2019-12-13:
> I get the feeling that the Creality board is fixed in standalone mode

You are correct.

stantond, 2019-12-14:
Having set the drivers to "TMC2208_STANDALONE", M569 still doesn't work. Is that expected?

stantond, 2019-12-14:
Tested K calibration print again, now on Marlin 2.0 with drivers set to TMC2208_STANDALONE, which is the only option for TMC drivers on the Creality/Melzi "Silent" boards v1.1.4 and v1.1.5.

Same result. Prints the first line where K=0, then extruder stepper stops, but print tries to continue as normal, unaware of the failure.

ManuelMcLure, 2019-12-14:
``M569`` depends on a UART or SPI connection to the driver to change stepping mode, so there's no way for it to work with a ''TMCXXXX_STANDALONE`` driver. I think if your 2208s are soldered to the board you're out of luck and will be unable to change them to SpreadCycle.

stantond, 2019-12-14:
The problem isn't M569 - it's just a means to an end. I'm trying to establish whether or not Linear Advance is compatible with StealthChop. At this point, I suspect not, and recommend the docs be updated, unless anyone else is able to further test this.

WireBe, 2019-12-16:
@stantond  I would try this:

  /**
   * Beta feature!
   * Create a 50/50 square wave step pulse optimal for stepper drivers.
   */
  #define SQUARE_WAVE_STEPPING

  /**
   * Enable M122 debugging command for TMC stepper drivers.
   * M122 S0/1 will enable continous reporting.
   */
  #define TMC_DEBUG

stantond, 2019-12-18:
> @stantond I would try this:
> 
> /**
> 
> * Beta feature!
> * Create a 50/50 square wave step pulse optimal for stepper drivers.
>   */
>   #define SQUARE_WAVE_STEPPING
> 
> /**
> 
> * Enable M122 debugging command for TMC stepper drivers.
> * M122 S0/1 will enable continous reporting.
>   */
>   #define TMC_DEBUG

Have just tried this. Thanks for the suggestion. Still didn't work - extruder stops extruding the moment it gets to the first bit where the K factor is changed to a non-zero value.

sjasonsmith, 2019-12-18:
@stantond I wonder if this is related to some pulse timing I observed yesterday related to Linear Advance. I posted a capture of step output on #16155.

I have seen waveforms similar to this cause 2208 drivers to shut down, although in that case it was a different issue not related to linear advance. 2209 drivers seemed more resilient when I encountered that earlier issue, but I assume they have a breaking point as well.

sjasonsmith, 2019-12-18:
It is also worth mentioning that SQUARE_WAVE_STEPPING won't do anything when you are using drivers in standalone mode. The image I posted to the other issue is with square wave, but a similar issue would exist without it.

boelle, 2019-12-24:
@stantond is the issue still the same with all the updates in the last 6 days?

stantond, 2019-12-24:
> @stantond is the issue still the same with all the updates in the last 6 days?

Updates? Have there been some bugfix releases since that could have affected this?

boelle, 2019-12-24:
that is what i'm asking you, i never had the problem so i cant see if there is any difference

stantond, 2019-12-24:
> that is what i'm asking you, i never had the problem so i cant see if there is any difference

Haven't tried it - don't usually update for each bugfix release, I thought the general advice was stick to the main releases. Are there actually any changes that look like they might have an impact, or is this more of a "just in case why not give it a go"?

boelle, 2019-12-24:
its "just in case why not give it a go" and if its not fixed go back again

stantond, 2019-12-24:
Ok, will try it in about a week when I'm back with the printer

Vertabreak, 2019-12-24:
video inbound of me running the calibration test on my machine with tmc2208 just fine its uploading.

stantond, 2019-12-24:
> video inbound of me running the calibration test on my machine with tmc2208 just fine its uploading.
> Ive run the calibration a few 100 times over the last year with the machine never once had the described issue.

What mode is your extruder driver in?

Vertabreak, 2019-12-24:
standalone same as yours.

stantond, 2019-12-24:
That sounds promising then, can you share your config files, please? I'll do a side-by-side to see if any differences could be causing this. Which mainboard are you using?

Vertabreak, 2019-12-24:
there on my github unfortunately my fork isn't gonna be easy to compare its one unified config supporting like 100 printers and there variants but it is kept updated daily with the latest marlin source.

boelle, 2019-12-24:
btw, who can confirm there is an issue? use the same configs as OP and the same hardware if possible

Vertabreak, 2019-12-24:
the 1284p is like the little brother to the 2560 it should be close enough to compare.

boelle, 2019-12-24:
i will then put a confirmed label on it

Vertabreak, 2019-12-24:
i totally disprove a problem exists if comparing them like this is acceptable.

boelle, 2019-12-24:
me misunderstod then.... label is gone and yes now that my brain is in gear i remember the issue

i never had it either so it must be a not confirmed

sjasonsmith, 2019-12-24:
I will try to reproduce it on my machine. I have been investigating LA causing 2208 drivers to turn off and am hoping to fix it. I’ve been reproducing with the default K value so haven’t actually tried the calibration print. It probably doesn’t happen to everyone due to other configuration settings, Vref settings, etc...

stantond, 2019-12-24:
> I will try to reproduce it on my machine. I have been investigating LA causing 2208 drivers to turn off and am hoping to fix it. I’ve been reproducing with the default K value so haven’t actually [...]

I set my default K to 0 in config, so I don't accidentally print with a value that doesn't work well for the filament type (and so that I can continue to print without it while troubleshooting this issue)

sjasonsmith, 2019-12-27:
After spending time looking into this and attempting various small patches to improve LA behavior on 2208 drivers, my recommendation is just to turn it off. In my opinion Linear Advance would have to be completely reworked to avoid the issues that cause the 2208 to shut down, and probably cause quality/noise issues with other drivers.

0n3-70uch, 2019-12-28:
Hi, I can confirm this problem with my Sovol SV01 and SilentBoard (TMC2208_STANDALONE).

Vertabreak, 2019-12-28:
its rather strange this only happens with some TMC2208 drivers as mine are entirely unaffected.

boelle, 2020-01-08:
can we do anything about this one when the error only happen on some TMC2208 ?

Fenixin, 2020-01-12:
It isn't very useful for the issue but I have exactly the same problem with Marlin bugfix-2.0.x. When Linear Advance is activated and k > 0 the extruder stops extruding. I have tried everything posted here and other things seen in other places, nothing seems to work. My motherboard is a Creality 3D 2.5.2 (picture attached just in case is useful).

[image]

The drivers are TMC2208 and as far as I know they are in standalone mode.

Watching closely as it prints the calibration pattern I can see that the stepper driver extrudes at the start of the first line with K>0, but its speed is reduced rapidly and it never regains movement (even when the k factor is set to zero at the end of the calibration pattern gcode).

If there is anything I can try/test/provide to help fix this just let me know.

pro100luk, 2020-01-12:
Confirming issue with my Ender 3 Pro and Creality silent bpard v1.1.15.
Today installed Marlin 2.0.1 with LA. Same result only k=0 line printed

boelle, 2020-01-12:
@pro100luk what with bugfix 2.0.x ?
2.0.1 is just a frozen copy of 2.0.x and is now "old"

pro100luk, 2020-01-13:
@boelle the same behavior with bugfix.
Also tried disabling STEALTHCHOP_E and it also didn't work.

sjasonsmith, 2020-01-13:
> After spending time looking into this and attempting various small patches to improve LA behavior on 2208 drivers, my recommendation is just to turn it off. In my opinion Linear Advance would have [...]

To expand on my earlier comment.

Linear advance inserts its own steps, out of sequence with the steps for normal movement. There is no synchronization between these two source of steps, which can cause you to perform two steps 1 microsecond apart, in the middle of a chain of steps 20 milliseconds apart. This rapid velocity shift appears to cause the 2208s to shut down, but may also cause rough operation with other drivers.

There is no combination of settings that will solve this problem, without reworking something to allow these two sources of steps to synchronize and accelerate smoothly.

If you are encountering this issue and aren't willing to perform that rewrite, you should probably just turn off Linear Advance.

stantond, 2020-01-13:
> To expand on my earlier comment.
>
> Linear advance inserts its own steps, out of sequence with the steps for normal movement. There is no synchronization between these two source of steps, which can [...]

So the LA implementation is somewhat flawed and could be causing low key wear or damage to systems where it appears to work?

sjasonsmith, 2020-01-13:
>  could be causing low key wear or damage to systems where it appears to work?

I can't speak to whether it could damage components. I suspect the most common issue (aside from 2208s shutting down) would be missed steps, but maybe it leads to problems involving increased temperatures, etc.

stantond, 2020-01-19:
It sounds like there are enough verifications now to keep this open. In the little time I have currently for 3D printing, I'm installing a SKR 1.4 with 2209s to go 32 bit and selling the Creality board, so it's unlikely I'll be able to further contribute to troubleshooting this issue (unless it occurs on this combination of board and drivers too, which also seems unlikely).

It the meantime, it might be worth adding a note in the Linear Advance documentation to highlight the current incompatibility with LA and Creality boards with 2208 drivers, linking to this issue.

macminoz, 2020-01-20:
I just want to leave here my own experience too...I also changed from Creality 1.1.3 to 1.1.5 and I encountered the same problems (I'm not an expert like you but the symptoms are exactly the same). I just reverted to 1.1.3 and seems everything back to works (Marlin 1.1.x). I'm using the E3D Hemera as a extruder.

stantond, 2020-01-20:
> I just want to leave here my own experience too...I also changed from Creality 1.1.3 to 1.1.5 and I encountered the same problems (I'm not an expert like you but the symptoms are exactly the same). [...]

That makes sense. Your 1.1.5 (and the "silent" 1.1.4) boards are using 2208 drivers, but I don't think (haven't checked) Creality have changed anything else since board 1.1.3, so the drivers are in standalone mode and stuck in stealthchop

[100 more replies]

closed 2022-07-21 by thisiskeithb