MarlinFirmware/Marlin #16078: [BUG] [HAL_STM32] sdcard eeprom emulation file corruption

Drachenkaetzchen, 2019-12-02:
# Bug Description

Prerequisites: A board using `HAL_STM32`, possibly affects `HAL_STM32F1` as well.

If `M500` fails due to a hardware or filesystem errors, the file descriptor stays open, causing subsequent writes to result in a 0 byte 'eeprom.dat' file but reports back a successful write.

In `PersistentStore::access_finish` it should also be checked for `-1` in `file.write`, as well as checking the return code for `PersistentStore::access_finish`.

### My Configurations

Not applicable.

### Additional information

This is mainly a reminder ticket for me to implement these fixes.

Drachenkaetzchen, 2019-12-03:
This issue depends on #16084

TerawattX, 2019-12-20:
I am running into this same issue, however I do not have EEPROM emulation enabled.

In my case I am running a SKR Mini E3 V1.2 set to use 512k of storage (firmware file is ~257kb compiled).  If I init the eeprom and only save basic values like probe offset, stepper mode, esteps/mm I can save and load the eeprom without issue.   However if I generate a UBL mesh and save it to either slot, then perform an M500, I get the version mismatch error when running M501.

sjasonsmith, 2019-12-20:
@TerawattX, this issue is specific to saving to an SD card, and mainly targeting a different HAL than you board uses. It sounds like you are probably storing to flash? If so I suspect you are hitting issue #15982, which was fixed about two weeks ago. The title specifies printcount, but UBL meshes caused the same problem.

TerawattX, 2019-12-22:
Thanks for the info!  Pretty sure its saving to flash, but I might try enabling saving to SD after taking a look at the link you provided.  

I'm not sure what's going on, but I definitely feel like I'm spending more time fighting to get the printer working than I am printing.  I've had print counts on with every build and have been able to save the UBL mesh without issue up until the other night.  

The only changes I've made to the printer lately was enabling (then disabling) a filament sensor, then turning on M48 and turning it back off.  Figure if one of those things caused the problem then turning them off would resolve it... unfortunately the SD card I was using died last night, so I lost all the old .bin files I had been hanging onto in case I ran into something like this. :\

Even weirder, I can generate the mesh, save it to S0, Load it from S0, then output it with G29 T and its fine.  I start a print and its gone.  The only thing Cura was doing before a print is resetting the BLTouch, then homing and loading the mesh.  I reload the mesh with 100x M421 commands and do the same thing again and its still there when the print starts.

sjasonsmith, 2020-07-04:
@boelle did you have a reason for removing the stale label from this one?

@Drachenkaetzchen or @TerawattX, are either of you still around and experiencing this problem?

thinkyhead, 2020-07-05:
The older labels are not relevant at this point, so it's fine if any were removed 5 days ago. As it happens, I removed all "stale" labels when I deleted the label itself (and re-added it) about 4 days ago.

The updated "close-stale" workflow will re-add the labels as soon as the action that it uses gets patched by the author (it is using some incompatible code and a PR is in the queue) — which could be soon or could be a while.

sjasonsmith, 2020-07-05:
I believe there is no reason for this to be open anymore. Flash-based EEPROM now works on this board, and I believe others are using SD successfully as well. Aside from the first batch there is even a physical EEPROM, which has it's own Feature Request open here: https://github.com/MarlinFirmware/Marlin/issues/18434

I am closing this, as it does not seem worthwhile to invest time in it. If anybody is still encountering issues involved in this, please let us know so it can be re-opened.

closed 2020-07-05 by sjasonsmith