MarlinFirmware/Marlin #16083: [BUG] The program size is greater than maximum allowed

TheZoker, 2019-12-03:
### Bug Description

Since the release of 2.0.0 the example config for the Ender 3 does not build anymore. 
Platform.io throws this error:
```
Linking .pio\build\melzi_optiboot\firmware.elf
Checking size .pio\build\melzi_optiboot\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
Error: The program size (123796 bytes) is greater than maximum allowed (64512 bytes)
DATA:    [=====*** [checkprogsize] Explicit exit, status 1
=====]  111.6% (used 4573 bytes from 4096 bytes)
PROGRAM: [==========]  191.9% (used 123796 bytes from 64512 bytes)
```

### My Configurations

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3917105/Marlin.zip)

### Steps to Reproduce

1. Download Marlin 2.0.0
2. Copy the examples from `config/examples/Creality/Ender-3` into `Marlin`
3. Modify `platform.ini` to use `melzi_optiboot` as the environment
4. Build 

**Expected behavior:** Build without issues

**Actual behavior:** Builds with issues

RudyBenner, 2019-12-03:
Like this?
[image]

RudyBenner, 2019-12-03:
same in bugfix2.0.x

thisiskeithb, 2019-12-03:
> PROGRAM: [==========]  191.9% (used 123796 bytes from 64512 bytes)

There should be 128k, not 64k, so PIO thinks you’re over the limit.

Under the `melzi_optiboot` environment, change: https://github.com/MarlinFirmware/Marlin/blob/0832a60db377a41c0aa8c6e81d256461b64cf2c4/platformio.ini#L145

To: `board = sanguino_atmega1284p` and recompile.

TheZoker, 2019-12-03:
Works, thanks!

thisiskeithb, 2019-12-03:
@boelle This is still a "bug" and should be looked at.

I know @thinkyhead has been doing some work on `platformio.ini`, but I'm not sure what the current plan is.

boelle, 2019-12-03:
oh just saw works thanks and hit close

thisiskeithb, 2019-12-03:
See https://github.com/MarlinFirmware/Marlin/pull/16089

closed 2019-12-04 by thinkyhead