MarlinFirmware/Marlin #16084: [BUG] SdBaseFile.open returns false yet opens the file

Drachenkaetzchen, 2019-12-03:
### Bug Description

If `file.open` gets called, it returns `false` despite the file has opened (or appears to be).

I'm currently trying to make SDCard eeprom emulation more reliable for STM32 platforms. Here's the code:

```c
if (file.isOpen()) {
    SERIAL_ECHOLNPGM("?EEPROM file already open. This is a bug.");
    return false;
}

if (!file.open(&root, EEPROM_FILENAME, O_CREAT | O_WRITE | O_TRUNC)) {
    if (file.isOpen()) {
      SERIAL_ECHOLNPGM("?open returned false but yet it is open");
    }
    SERIAL_ECHOLNPGM("?Unable to open EEPROM file for writing");
    return false;
}
```

As you can see, the method should return if the file is already open. However, `file.open` returns false yet ´file.isOpen()` returns `true` afterwards.

### My Configurations

Not applicable.

Drachenkaetzchen, 2019-12-03:
I've traced it down to `SdVolume::freeChain` where it returns false. As I'm non-familiar with the inner workings of FAT and SDCards, and it might be hard to fix, maybe a workaround would be to mark `O_TRUNC` as unsupported?

Right now, `O_TRUNC` is not widely used. The only occurences seem to be in the STM32 Sdcard emulation as well as with the binary protocol.

Drachenkaetzchen, 2019-12-03:
As a workaround for #16078 I simply delete the file if it exists, so there's no need for `O_TRUNC`. However, I don't know if the binary protocol is affected. I'll close the issue for now.

Drachenkaetzchen, 2019-12-03:
Unfortunately, deleting the file calls `truncate` internally which fails at `fatGet(cluster, &next))`.

tpruvot, 2019-12-08:
else, full size SD Cards often have a Read only switch, imo it could be interesting to inspect the behavior in this case...

Drachenkaetzchen, 2019-12-08:
> else, full size SD Cards often have a Read only switch, imo it could be interesting to inspect the behavior in this case...

The responsibility for the read only switch lies at the card reader and not the card itself. That "switch" is not electrically connected inside the sd card, so it's essentially a host software read only implementation.

boelle, 2020-02-08:
@Drachenkaetzchen i agree with you, sadly a lot of boards (mostly in the past) do not even connect the switch to anything, and i guess from that it has not bee thought of much in most 3d printer firmwares

but it should

boelle, 2020-02-08:
@thinkyhead any thoughts here?

thinkyhead, 2020-02-11:
The issue is interesting. I'm not sure of the implications for fast binary. Maybe none…? We would need to make a closer study of the code.

closed 2020-06-23 by thinkyhead