MarlinFirmware/Marlin #16086: [BUG] Layer skipping o MKS Sbase

Clint75, 2019-12-03:
I have just discoverd that there is a bug in MKS sbase drivers drv8825
I was doing a clean install on it and got small layershifts on all axis after some help on Facebook I did get som help to sort it out.
I did have to enable  in configuratiation_adv.h
#define MINIMUM_STEPPER_POST_DIR_DELAY 650
#define MINIMUM_STEPPER_PRE_DIR_DELAY 650
#define MINIMUM_STEPPER_PULSE 5
#define MAXIMUM_STEPPER_RATE 250000

and that sorted things out 
please uppdate the example config in the repository

sjasonsmith, 2019-12-03:
It is not obvious from the title, but this behavior is discussed in issue #11205 .

The settings listed here are a good workaround, but not a solution. There is a problem in Marlin causing it not to guarantee MINIMUM_STEPPER_PULSE. I am working on a fix, but it isn't quite ready to post yet. Maybe it will happen tonight.

boelle, 2019-12-04:
closing since this is most likely a dublicate of issue in #11205

we can reopen if its truly a different issue, we only need one open with the same issue

closed 2019-12-04 by boelle