MarlinFirmware/Marlin #16087: Error: "Please select no more than one LCD controller option." Wanhao D6

Deneteus, 2019-12-03:
**### Bug Description**

PlatformIO Build fails due to multiple errors when compiling using the Wanhao D6 Example configuration files included with Marlin 2.0 

**### My Configurations**

I used the stock example configuration files for the Wanhao D6 and they failed to compile.
\Marlin\config\examples\Wanhao\Duplicator 6\

### Steps to Reproduce

1. Copy the Wanhao D6 example configuration files to the Marlin project folder.
2. Open the Marlin project in Atom and confirm the configuration.h file is the correct file.
3. Under Platform IO Home choose Inspect and make sure the Project Environment is megatmega2560
4. Choose PlatformIO on the tool bar and select Build.
5.  Watch as the Sanity Check Error occurs and the build fails.

**Expected behavior:** Example Configuration should compile with no errors.

**Actual behavior:** Multiple sanity check errors occur related to multiple LCD controller options. 

#### Additional Information

[image]

[image]

thisiskeithb, 2019-12-04:
Both `ULTIPANEL` and `U8GLIB_SSD1306` are defined in that config. Comment out `ULTIPANEL` and see if your LCD still works after updating.

Deneteus, 2019-12-04:
> Both `ULTIPANEL` and `U8GLIB_SSD1306` are defined in that config. Comment out `ULTIPANEL` and see if your LCD still works after updating.

Disabling Ultipanel causes this: 

[image]

boelle, 2019-12-04:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

thisiskeithb, 2019-12-04:
> Configurations, please
> Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
> requested in the Issue template) and drop them into your next reply.
> We'll check them over and see if anything is amiss.

They are using the stock D6 configs:

> I used the stock example configuration files for the Wanhao D6 and they failed to compile.
\Marlin\config\examples\Wanhao\Duplicator 6

boelle, 2019-12-04:
> They are using the stock D6 configs:

did not read that, just saw a missing zip

Deneteus, 2019-12-04:
I figured there would be no point in attaching the stock files.
[Uploading Wanhao D6 Marlin 2.0 Stock Example Configs.zip…]()

boelle, 2019-12-04:
as said i did not read the post just looked if something was attached

but yeah standard config not working is kind of embarresing,

Deneteus, 2019-12-04:
I tested using the firmware as compiled after those two changes and it failed.

shitcreek, 2019-12-04:
The problem is line 1991: `// #define U8GLIB_SSD1306`
I've made the correction. Thanks.

closed 2019-12-04 by shitcreek

thisiskeithb, 2019-12-04:
Does this work with only `ULTIPANEL` defined? I thought `U8GLIB_SSD1306` was required since this printer has an OLED screen.

shitcreek, 2019-12-04:
It compiled without issue. @Deneteus mind testing this out to confirm it works with `U8GLIB_SSD1306` undefined?

thisiskeithb, 2019-12-04:
Digging back into the origins of this config, it came with both those LCD types enabled from the start. I wonder if something change in `SanityCheck.h` for this to start throwing errors or if this config ever worked in the first place.

Deneteus, 2019-12-05:
@shitcreek I had not tested it with the SSD1306 OLED LCD undefined.
@thisiskeithb  The motherboard for the D6 is an Ultimaker 2 clone. It is based on the ATMega2560 microcontroller. 

[image]

thisiskeithb, 2019-12-05:
If that doesn't work, it seems this [Marlin port]( https://github.com/dot-bob/Marlin-Duplicator-6) is popular and they have a different LCD defined:

```cpp
//
// TinyBoy2 128x64 OLED / Encoder Panel
//
#define OLED_PANEL_TINYBOY2
#define LCD_RESET_PIN 5
```

Deneteus, 2019-12-05:
> If that doesn't work, it seems this [Marlin port](https://github.com/dot-bob/Marlin-Duplicator-6) is popular and they have a different LCD defined:
> 
> ```c++
> //
> // TinyBoy2 128x64 OLED / Encoder Panel
> //
> #define OLED_PANEL_TINYBOY2
> #define LCD_RESET_PIN 5
> ```

I'll test that in a minute!

thisiskeithb, 2019-12-05:
This config builds with a single LCD defined now (as expected), so we just need confirmation that it works. 😄

Here's the binary I built while testing with `TINYBOY2` enabled: [Wanhao-D6-Marlin-2.0.x-f646499.hex.zip](https://github.com/MarlinFirmware/Marlin/files/3925303/Wanhao-D6-Marlin-2.0.x-f646499.hex.zip)

Deneteus, 2019-12-05:
Got it! Just gonna go flash it!

Deneteus, 2019-12-05:
I am waiting on a netbook that I use to update firmware to finish restarting from a Windows Update. It  had been going for almost 8hrs. LOL

Deneteus, 2019-12-05:
Alright it flashed. LCD works. I just need to build it with the Tiny LCD settings and different settings for a Bondtech extruder.

thisiskeithb, 2019-12-05:
Thanks for finding this bug 🙂

PR #16117 will fix the default D6 config.

Deneteus, 2019-12-05:
Do you know if there is an issue related to this? 

[image]

Deneteus, 2019-12-05:
I got my printer updated with all of the changes I needed. Rob's config files had some tweaks to the default feed rates, adaptive step smoothing and power loss recovery. 

I enabled S_CURVE_acceleration, Bondtech ESTEPS, Tuned A4988 Stepper Defaults, Raised the max temp since I upgraded to Microswiss All Metal.

[Deneteus Marlin 2.0 D6 Configs.zip](https://github.com/MarlinFirmware/Marlin/files/3925776/Deneteus.Marlin.2.0.D6.Configs.zip)

shitcreek, 2019-12-05:
`unused variable` warnings can be ignored. Ideally we should have`isHeat`nested inside a condition.