MarlinFirmware/Marlin #16088: [BUG] G29 S-1 doesn't output any gcode after G29 P1 has completed

Drachenkaetzchen, 2019-12-03:
### Bug Description

When doing a `G29 S-1` on factory defaults, lots of gcode commands are written to the console.

After a `G29 P1` has completed, `G29 S-1` does not output any gcode.

### My Configurations

[Configuration.h](https://github.com/Drachenkaetzchen/Marlin/blob/PurpleCube-S6/Marlin/Configuration.h)
[Configuration_adv.h](https://github.com/Drachenkaetzchen/Marlin/blob/PurpleCube-S6/Marlin/Configuration_adv.h)

### Steps to Reproduce

1. Run `G29 P1`
2. Run `G29 S-1`

**Expected behavior:**

Actual mesh should be outputted.

**Actual behavior:**

No output.

randellhodges, 2019-12-03:
I can confirm. (I can only probe 43 out of 49 points, those 7 are the cause of the problem) 

If you toss in a `G29 P3` before the `G29 S-1`, it'll work. A quick peek at the code reveals a call to `mesh_is_valid`, which if any point isnan, it won't export anything.

Drachenkaetzchen, 2019-12-03:
I thought I had done a `G29 P3` here as well and received also no output. I'll verify that shortly.

tatusah, 2019-12-03:
+1 here, without filling the incomplete mesh with `G29 P3` the `G29 S-1` doesn't output the mesh in gcode. SKR mini e3 v1.2

Drachenkaetzchen, 2019-12-03:
`G29 P3` does not work after a `G29 P1`.

Result:
```
> G29 P1
Probing mesh point 1/100.
[...]
Probing mesh point 49/100.
X:150.00 Y:60.00 Z:20.00 E:0.00 Count A:16800 B:7200 Z:8000
ok
> G29 S-1
ok
> G29 P3
ok
> G29 S-1
ok
```

Drachenkaetzchen, 2019-12-03:
A `G29 T` results in:

```
Send: G29 T
Recv: 
Recv: Bed Topography Report:
Recv: 
Recv:     (  1,279)                                                              (299,279)
Recv:         0       1       2       3       4       5       6       7       8       9
Recv:  9 | -0.000  -0.036  -0.072  -0.022  -0.030  -0.026  -0.025  +0.009  +0.012  +0.014
Recv:    |
Recv:  8 | -0.062  -0.067  -0.072  -0.034  -0.050  -0.039  -0.025  +0.000  -0.012  -0.012
Recv:    |
Recv:  7 | -0.098  -0.098  -0.064  -0.045  -0.071  -0.051  -0.007  -0.008  -0.036  -0.036
Recv:    |
Recv:  6 | -0.072  -0.078  -0.083  -0.023  -0.010  -0.023  -0.032  +0.002  -0.061  -0.061
Recv:    |
Recv:  5 | -0.069  -0.069  -0.053  -0.019  +0.002  +0.002  +0.002  -0.047  -0.063  -0.063
Recv:    |
Recv:  4 | -0.103  -0.103  -0.057  -0.079  -0.024 [-0.035] -0.040  -0.057  -0.062  -0.062
Recv:    |
Recv:  3 | -0.104  -0.104  -0.098  -0.047  -0.033  -0.074  -0.106  -0.110  -0.085  -0.059
Recv:    |
Recv:  2 | -0.104  -0.104  -0.098  -0.015  -0.033  -0.074  -0.106  -0.110  -0.085  -0.059
Recv:    |
Recv:  1 |   .       .       .       .       .       .       .       .       .       .
Recv:    |
Recv:  0 |   .       .       .       .       .       .       .       .       .       .
Recv:         0       1       2       3       4       5       6       7       8       9
Recv:     (  1,  1)                                                              (299,  1)
Recv: 
```

It seems that `G29 P3` doesn't fill in some points, and the user is never notified about that.

Drachenkaetzchen, 2019-12-03:
I tried to use the mesh editor to fill in the remaining values, but the bed crashed into the nozzle as Marlin didn't seem to use either BLTouch as an endstop or was confused about the mesh value being invalid.

randellhodges, 2019-12-03:
Give it a try with a 7x7 grid. That is what I used on an SKR 1.3. I end up with only 1 set of points that cannot be probed due to my probe offset and the P3 works.

tatusah, 2019-12-03:
I think that you have to use `G29 P3` as many times as you have empty rows or columns in the mesh after the P1. Like in my case using 9x9 and the edge / offset limits I use there are 2 empty columns and I've to use G29 P3 twice to fill those both. This is the way I've seen the smart fill work for the time I've been using it.

Drachenkaetzchen, 2019-12-03:
I think that `G29 S-1` should report back the values, even if the mesh is invalid. This would be consistent with `G29 Sx`.

randellhodges, 2019-12-03:
I would think that should be safe. Just need to remove one line.

https://github.com/MarlinFirmware/Marlin/blob/7e7cbecf9ae001d40f6d9398fc3105a89716ecaa/Marlin/src/feature/bedlevel/ubl/ubl.cpp#L47

Would there be any issues running the M421 it spits out but have some data "lingering" in the previous nan points? Or do you think it should also export the nan state and have M421 import that as well?

Drachenkaetzchen, 2019-12-03:
I think it should behave like `G29 Sn` to have it consistent. The reason I'm asking for this is since flash is currently broken on the Fysetc S6 / STM32, and sdcard emulation isn't working either, I'm looking for a way to at least manually load/save UBL states via gcode.

phongshader, 2020-06-26:
has this been fixed, I've been unable to run any G29 command with the exception of G29 T

closed 2020-08-01 by github-actions[bot]