MarlinFirmware/Marlin #16091: [BUG] TMC2209 Sensorless Homing Not Working

elementfoundry, 2019-12-03:
Hardware: Bigtreetech SKR 1.3, Bigtreetech 2209 V1.2

I've tried enabling sensorless homing with my setup, but the "endstops" never trigger.

Debug info (this was captured during a G28 home all axis)

Send: M122 S1
Recv: axis:pwm_scale/curr_scale/mech_load|flags|warncount
Recv: 		X	Y	Z	E
Recv: Address		0	0	0	0
Recv: Enabled		false	false	false	false
Recv: Set current	1000	1000	1000	1000
Recv: RMS current	994	994	994	994
Recv: MAX current	1402	1402	1402	1402
Recv: Run current	17/31	17/31	17/31	17/31
Recv: Hold current	8/31	8/31	8/31	8/31
Recv: CS actual	8/31	8/31	8/31	8/31
Recv: PWM scale	10	10	10	10
Recv: vsense		0=.325	0=.325	0=.325	0=.325
Recv: stealthChop	false	false	false	false
Recv: msteps		16	16	16	16
Recv: tstep		max	max	max	max
Recv: pwm
Recv: threshold
Recv: [mm/s]
Recv: OT prewarn	false	false	false	false
Recv: OT prewarn has
Recv: been triggered	false	false	false	false
Recv: off time	4	4	4	4
Recv: blank time	24	24	24	24
Recv: hysteresis
Recv: -end		2	2	2	2
Recv: -start		1	1	1	1
Recv: Stallguard thrs	128	128	0	0
Recv: DRVSTATUS	X	Y	Z	E
Recv: stst		*	*	*	*
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: 157C
Recv: 150C
Recv: 143C
Recv: 120C
Recv: s2vsa
Recv: s2vsb
Recv: Driver registers:
Recv: 		X	0x80:08:00:00
Recv: 		Y	0x80:08:00:00
Recv: 		Z	0x80:08:00:00
Recv: 		E	0x80:08:00:00
Recv: 
Recv: 
Recv: Testing X connection... OK
Recv: Testing Y connection... OK
Recv: Testing Z connection... OK
Recv: Testing E connection... OK
[...]
Send: G28
Recv: X:10/8/-|I|	Y:10/8/-|I|	Z:27/17/-||	E:10/8/-|I|
Recv: X:10/8/-|I|	Y:10/8/-|I|	Z:27/17/-||	E:10/8/-|I|
[...]
Recv: X:10/8/-|I|	Y:10/8/-|I|	Z:20/17/-||	E:10/8/-|I|
Recv: X:44/17/-||	Y:10/8/-|I|	Z:20/17/-|I|	E:10/8/-|I|
Recv: echo:busy: processing
Printer seems to support the busy protocol, will adjust timeouts and set busy interval accordingly
Recv: X:44/17/-||	Y:10/8/-|I|	Z:20/17/-|I|	E:10/8/-|I|
Recv: X:43/17/-||	Y:10/8/-|I|	Z:20/17/-|I|	E:10/8/-|I|
[...]
Recv: X:44/17/-||	Y:10/8/-|I|	Z:20/17/-|I|	E:10/8/-|I|
Recv: X:43/17/-||	Y:10/8/-|I|	Z:20/17/-|I|	E:10/8/-|I|
Recv: echo:busy: processing
Recv: X:44/17/-||	Y:10/8/-|I|	Z:20/17/-|I|	E:10/8/-|I|
Recv: X:44/17/-||	Y:10/8/-|I|	Z:18/15/-|I|	E:10/8/-|I|
[...]
Recv: X:43/17/-||	Y:10/8/-|I|	Z:14/12/-|I|	E:10/8/-|I|
Recv: X:44/17/-||	Y:10/8/-|I|	Z:12/10/-|I|	E:10/8/-|I|
Recv: echo:busy: processing
Recv: X:44/17/-||	Y:10/8/-|I|	Z:10/8/-|I|	E:10/8/-|I|
Recv: X:44/17/-||	Y:10/8/-|I|	Z:10/8/-|I|	E:10/8/-|I|
[...]
Recv: X:44/17/-||	Y:10/8/-|I|	Z:10/8/-|I|	E:10/8/-|I|
Recv: X:44/17/-||	Y:10/8/-|I|	Z:10/8/-|I|	E:10/8/-|I|
Recv: echo:busy: processing
Recv: X:43/17/-||	Y:10/8/-|I|	Z:10/8/-|I|	E:10/8/-|I|
Recv: X:44/17/-||	Y:10/8/-|I|	Z:10/8/-|I|	E:10/8/-|I|
[...]
Recv: X:43/17/-||	Y:10/8/-|I|	Z:10/8/-|I|	E:10/8/-|I|
Recv: X:32/17/-||	Y:10/8/-|I|	Z:10/8/-|I|	E:10/8/-|I|
Recv: Error:Printer halted. kill() called!

hammerhead2k2, 2019-12-03:
I think it is necessary to activate Stealth Chop for sensorless homing.
Regarding your log, i think it is disabled.

> Recv: stealthChop false false false false

See also issue  #15424

elementfoundry, 2019-12-03:
unless I'm missing something it is enabled in Configuration_adv.h

  #define STEALTHCHOP_XY
  #define STEALTHCHOP_Z
  #define STEALTHCHOP_E

opiswahn, 2019-12-04:
Had the same problem on my SKR E3 Mini V1.2... StealthChop is enabled. No triggering of 'endstops' on hit whatsoever.
Any advice would be helpful.

hammerhead2k2, 2019-12-04:
Did you double check the wiring, jumpers and settings?

Diag Pin at the driver has to be connected to the Board.

This one is from TMC2209 V1.2 Bigtreetech
[image]

**Unused Diag Pins at Bigtreetech Drivers (maybe E and Z) has to be removed! Otherwise you get in trouble with the endstops! I don't know, if there is an issue with Watterott or Fysetc too.** 

Fysetc V3.0 seems to have same pinning as BTT
[image]

This one is from Watterott and the pinning seems to be different.
[image]

Connected to the Mainboard here:
[image]

And activated the jumpers for X and Y Endstops over the DIAG Pin. The other ones have to be removed.
[image]

Disconnect the mechanical endstops:
[image]

In Configuration.h

`#define X_MIN_ENDSTOP_INVERTING true // Set to true to invert the logic of the endstop.`
`#define Y_MIN_ENDSTOP_INVERTING true // Set to true to invert the logic of the endstop.`

In Configuration_adv.h

` #define SENSORLESS_HOMING // StallGuard capable drivers only`

elementfoundry, 2019-12-04:
Upon further investigation it would appear that Marlin is not putting the tmc2209 drivers into stealthchop mode before moving the axes that require them before homing. Setting stealthchop manually before homing seems to work. So I think the solution to this problem is for marlin to check for stealthchop and enable if necessary before any sensorless homing.

boelle, 2019-12-07:
@teemuatlut are @elementfoundry on to something  here?

boelle, 2019-12-07:
@elementfoundry since 2.0 was just released a few days ago has this changed this issue at all?

elementfoundry, 2019-12-08:
@boelle - this was using marlin 2.0

uorbe001, 2019-12-14:
> Upon further investigation it would appear that Marlin is not putting the tmc2209 drivers into stealthchop mode before moving the axes that require them before homing. Setting stealthchop manually [...]

@elementfoundry How did you come to that conclusion? any way to test the theory? 

I'm having a similar issue (https://github.com/MarlinFirmware/Marlin/issues/15848) but in my case sensorless homing always triggers (as soon as it starts) with a threshold value (with my current config 160) but it never does so when I set the number lower than that (159). As far as I can tell I do have stealhchop enabled when it homes:

```
Driver stepping mode:
M569 S1 X Y Z
M569 S1 T0 E
M569 S1 T1 E
````

But there may something I'm missing?

uorbe001, 2019-12-14:
I saw your PR (https://github.com/MarlinFirmware/Marlin/pull/16153) and tried flashing that to test if it changed anything for me, but no luck, so I'm afraid mine is unrelated 😞

elementfoundry, 2019-12-15:
@uorbe001 : i came to that conclusion due to the line:

` * TMC2209 requires STEALTHCHOP enabled for SENSORLESS_HOMING`

in configuration_adv.h, the comment from @hammerhead2k2 at the beginning of the thread, and observing behavior on my printer when manually putting the drivers into stealthchop before homing with sensorless homing turned on.

in your case, it sounds like its a sensitivity issue, which was different from what I had written the patch for.

boelle, 2019-12-24:
btw, who can confirm there is an issue? use the same configs as OP and the same hardware if possible

elementfoundry, 2019-12-28:
@boelle - still seems to be an issue with 2.0.1 - sensorless homing fails to trigger when set to spreadcycle with tmc2209 drivers

Grogyan, 2019-12-28:
TMC2209 requires Stealth Mode for sensorless homing

boelle, 2020-01-07:
@elementfoundry is this still a problem?

boelle, 2020-01-14:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-01-14 by boelle