MarlinFirmware/Marlin #16092: [BUG] compilin error

ohnide, 2019-12-03:
### Bug Description

Hi everyone, new player here, 
I have been try to flash an MKS GEN L board for my tevo tarantula pro and wile compiling, gives this error

Arduino: 1.8.10 (Windows 10), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

C:\Users\Edgar Varela\Documents\Arduino\libraries\U8glib\utility\u8g_rot.c:48:27: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]

 u8g_dev_t u8g_dev_rot = { u8g_dev_rot_dummy_fn, NULL, NULL };

                           ^~~~~~~~~~~~~~~~~~~~

C:\Users\Edgar Varela\Documents\Arduino\libraries\U8glib\utility\u8g_rot.c:48:27: note: (near initialization for 'u8g_dev_rot.dev_fn')

avr-gcc: error: CreateProcess: No such file or directory

Multiple libraries were found for "U8glib.h"
 Used: C:\Users\Edgar
Multiple libraries were found for "TMCStepper.h"
 Used: C:\Users\Edgar
Multiple libraries were found for "SPI.h"
 Used: C:\Users\Edgar
Multiple libraries were found for "SoftwareSerial.h"
 Used: C:\Users\Edgar
exit status 1
Error compiling for board Arduino Mega or Mega 2560.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

### My Configurations

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3918642/Configuration.zip)

Grogyan, 2019-12-03:
Please try with PlatformIO for VSCode

thisiskeithb, 2019-12-03:
You'll need to install [Arduino 1.8.8 or newer](https://www.arduino.cc/en/Main/Software#download) to [build Marlin 2.0](https://github.com/MarlinFirmware/Marlin/tree/2.0.x#building-marlin-20) or as @Grogyan mentioned, use VSCode & PlatformIO.

ellensp, 2019-12-04:
The provided configuration files compile fine with latest marlin bugfix 2.0.x under Arduino 1.8.10. The only change I had to make was to comment out SHOW_CUSTOM_BOOTSCREEN since there is no Marlin/_Bootscreen.h by default.  I suspect this is an issue with the users build environment.

closed 2019-12-04 by boelle