MarlinFirmware/Marlin #16094: The terminal process terminated with exit code: 1 - SKR 1.3 Marlin 2.0, can't seem to upload

JAdams12, 2019-12-04:
Hi all, I'm having a problem getting Marlin to upload to my brand new SKR 1.3 board. I tried both Atom and Visual Studio, Atom gave me even more errors so I'm trying Visual Studio for now. 

This is the attempt at a "test" upload of bone stock Marlin, the only changes I did was change the MOTHERBOARD and SERIAL connections in configuration.h (serial connection one is now -1, and two is now 0) and changed the processor in platformio.ini to the LPC1768. 

Can anyone help? I'm having trouble finding fixes when googling, as issues are not as well documented as it was for Marlin 1.x.x

Thank you so much, below is my log

```
Processing LPC1768 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.2.zip; board: nxp_lpc1768; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1768.html
PLATFORM: NXP Arduino LPC176x 0.1.1 > NXP LPC1768
HARDWARE: LPC1768 100MHz, 31.97KB RAM, 464KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink)
PACKAGES: toolchain-gccarmnoneeabi 1.80201.190214 (8.2.1), framework-arduino-lpc176x 0.2.2
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ strict
Found 6 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Servo> 1.0.0
|-- <LiquidCrystal> 1.0.0
|-- <U8glib-HAL> 0.4.1
|-- <TMCStepper> 0.6.1
|-- <Adafruit NeoPixel> 1.3.0
|-- <SailfishLCD>
Building in release mode

Unable to find destination disk (Autodetect Error)
Please select it in platformio.ini using the upload_port keyword (https://docs.platformio.org/en/latest/projectconf/section_env_upload.html) or copy the firmware (.pio/build/LPC1768/firmware.bin) manually to the appropriate disk

Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\DebugMonitor.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\HAL.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\HAL_SPI.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\MarlinSerial.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\fast_pwm.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\include\digipot_mcp4451_I2C_routines.c.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\main.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\persistent_store_flash.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\persistent_store_sdcard.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\timers.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\u8g\u8g_com_HAL_LPC1768_hw_spi.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\u8g\u8g_com_HAL_LPC1768_ssd_hw_i2c.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\u8g\u8g_com_HAL_LPC1768_st7920_hw_spi.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\u8g\u8g_com_HAL_LPC1768_st7920_sw_spi.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\u8g\u8g_com_HAL_LPC1768_sw_spi.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\usb_serial.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\watchdog.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\shared\HAL_spi_L6470.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\shared\backtrace\backtrace.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\shared\backtrace\unwmemaccess.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\shared\eeprom_i2c.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\shared\eeprom_spi.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\shared\persistent_store_api.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\shared\servo.cpp.o
Compiling .pio\build\LPC1768\src\src\Marlin.cpp.o
Compiling .pio\build\LPC1768\src\src\core\serial.cpp.o
Compiling .pio\build\LPC1768\src\src\core\utility.cpp.o
Compiling .pio\build\LPC1768\src\src\feature\I2CPositionEncoder.cpp.o
Compiling .pio\build\LPC1768\src\src\feature\Max7219_Debug_LEDs.cpp.o
Compiling .pio\build\LPC1768\src\src\feature\babystep.cpp.o
Compiling .pio\build\LPC1768\src\src\feature\backlash.cpp.o
Compiling .pio\build\LPC1768\src\src\feature\baricuda.cpp.o
Compiling .pio\build\LPC1768\src\src\feature\bedlevel\abl\abl.cpp.o
Compiling .pio\build\LPC1768\src\src\feature\bedlevel\bedlevel.cpp.o
Compiling .pio\build\LPC1768\src\src\feature\bedlevel\mbl\mesh_bed_leveling.cpp.o
Compiling .pio\build\LPC1768\src\src\feature\bedlevel\ubl\ubl.cpp.o
[... 394 lines cut ...]
STM32F7                       IGNORED
ARMED                         IGNORED
STM32F103VE_longer            IGNORED
mks_robin                     IGNORED
mks_robin_lite                IGNORED
mks_robin_mini                IGNORED
mks_robin_nano                IGNORED
jgaurora_a5s_a1               IGNORED
STM32F103CB_malyan            IGNORED
chitu_f103                    IGNORED
FYSETC_S6                     IGNORED
STM32F407VE_black             IGNORED
BIGTREE_SKR_PRO               IGNORED
BIGTREE_BTT002                IGNORED
teensy31                      IGNORED
teensy35                      IGNORED
esp32                         IGNORED
linux_native                  IGNORED
SAMD51_grandcentral_m4        IGNORED
include_tree                  IGNORED
============================================================================================= 1 failed, 0 succeeded in 00:02:07.241 =============================================================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.
```

JAdams12, 2019-12-04:
I decided to try remote upload to the LPC1768 and it wrote a file to the SD card that then had the firmware on it. Restarted and it's working now. 

So for those of you who may have this issue, in Visual Studio, hit the cloud with the arrow at the bottom blue bar and look for Platformio: Upload (LC1768) and it should write it to the MicroSD card included with your printer.

closed 2019-12-04 by JAdams12

ellensp, 2019-12-04:
The SD card needs to be labeled REARM or have a file FIRMWARE.CUR in the root directory for the script to be able to locate what drive to upload the firmware.bin onto

p3p, 2019-12-04:
> Unable to find destination disk (Autodetect Error)
Please select it in platformio.ini using the upload_port keyword (https://docs.platformio.org/en/latest/projectconf/section_env_upload.html) or copy the firmware (.pio/build/LPC1768/firmware.bin) manually to the appropriate disk

As long as the MKS 1.3 is connected to the computer and `FIRMWARE.CUR` is on the sdcard the upload script should be able to detect it, not sure why the upload remote button worked as it will just be running the same script.

> The SD card needs to be labeled REARM

This isn't correct anymore, I think it was for a short time when the script was first added though, just the presence of `FIRMWARE.CUR` is required.

ellensp, 2019-12-04:
Current script https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/HAL/HAL_LPC1768/upload_extra_script.py still looks for REARM

p3p, 2019-12-04:
Ah yes sorry, I must have second guessed removing that check even though the name makes no sense, so you were indeed correct either `labeled REARM or have a file FIRMWARE.CUR in the root directory`.

ellensp, 2019-12-04:
The student has surpassed the master (on this one tiny instance :) ) I agree it no longer makes sense as its not just for rearm boards anymore but I think it should be be maintained as legacy as lots of documentation mentions this. But perhaps adding a "MARLIN2" label check would be a good idea and update the documentation over time to reflect this.

JAdams12, 2019-12-04:
Interesting. My understanding was that the IDE actually flashed the firmware like the AVR boards did, but now I get that it is totally reliant on the SD card. I did end up doing the Nintendo cartridge blow maneuver to the SD card slot before I hit the remote upload, maybe that's what fixed the issue.

I just tried the "regular" upload method and it worked perfectly.

So can I not use the onboard SD card for anything other than the firmware file? I was under the impression that I could use it for gcode. Not that it matters in this instance as I'm using a display with SD Card reader, but I bought a second one I was gonna use with a display I have that doesn't have an SD reader

Thank you both for the clarifications, I feel like a fish out of water with all this 32bit stuff, but I'm excited!

ellensp, 2019-12-04:
Yes you can also use the on board sd card for gcode.

p3p, 2019-12-04:
You can use the onboard card for gcode, the standard use case is just putting the files on the card over usb then printing, but the computer and Marlin can not access the card at the same time so you will need to go into the menu on the display and unmount it before it becomes available as a usb drive.

You can to choose how you want to use Marlin in the configuration, either with the display sd card which can be unreliable or slow over the long ribbon cables but easy to change, or the onboard card that is accessible over usb but harder to physically access.

JAdams12, 2019-12-04:
Oh wow, that's awesome! Do I want to do anything besides uncomment enable SD support under the extra settings part of config.h? 

I designed this printer so that the electronics box is easy to access with the SD card slow facing up, so accessing that will be just as easy as accessing the one on the display