MarlinFirmware/Marlin #16096: [BUG] (short description)

DroneMang, 2019-12-04:
Hi,

I have an SKR PRO.  I am trying to use one of the 3 fan outputs to control the hotend cooling fan.  I am using the other two for the part cooling fan and a case cooling fan.  I have a 2-1 hotend.  If I put FAN1_PIN for E0 and E1 or just for E0, the build in platformio using Atom fails.  If I put the actual pin from the pins file it also fails.  The uploaded zip produces the error.  

Also, a request to clarify the documentation in the code, I am wanting to use a 12v fan for cooling the hotend.  If I run the board on 24v, but hook the 12v fan to FAN1_PIN and set max pwm to 128, will the board supply the fan with 12v?

Thank you!
[Marlin-bugfix-2.0.x-6-broken-for-upload.zip](https://github.com/MarlinFirmware/Marlin/files/3919377/Marlin-bugfix-2.0.x-6-broken-for-upload.zip)

ellensp, 2019-12-04:
" If I run the board on 24v, but hook the 12v fan to FAN1_PIN and set max pwm to 128, will the board supply the fan with 12v?"  No It will supply the fan with pulse of 24v.  You need to consider wattage. Going from 12v to 24v with the same load you get 4 times the wattage.  You need max pwm set to 64 to for the same wattage.This assumes the 24v pulse doesn't kill it.

DroneMang, 2019-12-04:
Ok. I'll just run the 12v fan off of the 12v pin on the SKR PRO. But,
perhaps you could add what you just said to config_adv to what it already
says there? Thanks.

ellensp, 2019-12-04:
Th error is error: non-constant condition for static assertion in the line static_assert(PWM_PIN(E0_AUTO_FAN_PIN), "E0" AF_ERR_SUFF);  in SanityCheck.h  
It really hates PWM_PIN()

ellensp, 2019-12-04:
This seems to be he same error as #15362

ellensp, 2019-12-04:
As a work around change line 1868 of  SanityCheck.h from static_assert(PWM_PIN(E0_AUTO_FAN_PIN), "E0" AF_ERR_SUFF); to static_assert(true, "E0" AF_ERR_SUFF);   Which tells the compiler yes your E0_AUTO_FAN_PIN is PWM capable always (even if it isn't).

boelle, 2019-12-04:
most likely a dublicate so closing, we only need one report of same  issue

can reopen if its truly a different issue

closed 2019-12-04 by boelle