MarlinFirmware/Marlin #16097: [BUG] PlatformIO.INI - Incorrect monitor_speed sanguino_atmega1284p

Deneteus, 2019-12-04:
### Bug Description

When inspecting PlatformIO.ini to change the default default_envs for a Creality CR-10 build I discovered that the Sanguino Atmega 1284p board has the incorrect monitor_speed. This info is also  incorrect for the rest of the Melzi profiles. 

[env:megaatmega1280]
[env:sanguino_atmega1284p]
[env:melzi]

### Steps to Reproduce

1.  Open the platformio.ini file in Atom.
https://github.com/MarlinFirmware/Marlin/blob/2.0.x/platformio.ini

2.  Goto the sections below and verify the **monitor_speed** 

_[env:megaatmega1280]
[env:sanguino_atmega1284p]_

**Expected behavior:** I would expect the default baud rate to be set to 115200 or less for Sanguino and Atmega 1280 boards. I can't reliably connect to any of the Sanguino Creality boards at 250000.

**Actual behavior:** The monitor_speed is set to 250000

#### Additional Information

[image]

**https://github.com/Lauszus/sanguino/ **

[image]
****

shitcreek, 2019-12-04:
The `monitor_speed` is for platformIO's serial monitor to communicate with your board. You should change it to match the baud rate you've set in your configuration file to talk with it - basically in lewd of an interface such as that of Arduino's serial monitor with its dropdown for changing baud rate, it is set by the text file ini.
The reason 250000 is the default is because it has the least jitter. However there are many things that can affect the communication such poorly insulated wires. 
This isn't a bug as it is stated in configuration file that user must set the baud rate to what works best for them.

thisiskeithb, 2019-12-04:
I have a PR in to fix part of this issue: https://github.com/MarlinFirmware/Marlin/pull/16090

shitcreek, 2019-12-04:
dang it @thisiskeithb always undermining me, haha. good work though

Deneteus, 2019-12-04:
> The `monitor_speed` is for platformIO's serial monitor to communicate with your board. You should change it to match the baud rate you've set in your configuration file to talk with it - basically [...]

I actually was going to post a AVR Error rate chart in regards to this but I decided it's pretty obvious depending on the Mhz the more errors there are.

boelle, 2019-12-04:
the pr is merged so will close this one 

we can always reopen

closed 2019-12-04 by boelle