MarlinFirmware/Marlin #16101: [BUG] Software endstops ignore M206 settings

karabas2011, 2019-12-04:
### Bug Description

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

Home to Z max
M114 shows Zmax
M206 Zxxx add offset
G28 Z
G1 Z0
M114 shows Zxxx insteed of 0

**Expected behavior:** [What you expect to happen]
M114 shows Z0

**Actual behavior:** [What actually happens]
M114 shows Zxxx insteed of 0

#### Additional Information
Currently workaround is to turn off software endstop Zmax

ellensp, 2019-12-04:
the axis length does not change. you add xxx to Zmax, then Zmin becomes xxx

boelle, 2019-12-07:
will close this one until requirements in the issue template are satisfied

closed 2019-12-07 by boelle