MarlinFirmware/Marlin #16104: [BUG] STM32 compile error with TX_BUFFER_SIZE > 0

randellhodges, 2019-12-04:
### Bug Description

Build error with STM32 and TX_BUFFER_SIZE > 0

```
Compiling .pio\build\BIGTREE_SKR_PRO\src\src\feature\controllerfan.cpp.o
In file included from c:\code\public\marlin\marlin\src\inc\marlinconfig.h:45:0,
                 from Marlin\src\feature\bedlevel\ubl\ubl.cpp:23:
Marlin\src\feature\bedlevel\ubl\ubl.cpp: In static member function 'static void unified_bed_leveling::display_map(int)':
c:\code\public\marlin\marlin\src\core\serial.h:81:44: error: 'class USBSerial' has no member named 'flushTX'; did you mean 'flush'?
   #define SERIAL_FLUSHTX()      SERIAL_OUT(flushTX)
                                            ^
c:\code\public\marlin\marlin\src\core\serial.h:62:50: note: in definition of macro 'SERIAL_OUT'
   #define SERIAL_OUT(WHAT, V...) (void)MYSERIAL0.WHAT(V)
                                                  ^~~~
Marlin\src\feature\bedlevel\ubl\ubl.cpp:214:9: note: in expansion of macro 'SERIAL_FLUSHTX'
         SERIAL_FLUSHTX();
         ^~~~~~~~~~~~~~
*** [.pio\build\BIGTREE_SKR_PRO\src\src\feature\bedlevel\ubl\ubl.cpp.o] Error 1
```

### My Configurations

https://github.com/randellhodges/Marlin/tree/skr-1.1-pro/Marlin

### Steps to Reproduce

1. Build a SKR_PRO_1_1
2. Set TX_BUFFER_SIZE to something larger than 0

**Expected behavior:** To compile

**Actual behavior:** Doesn't compile

#### Additional Information

From serial.h:

```
#ifdef __STM32F1__
  #define SERIAL_FLUSHTX()      SERIAL_OUT(flush)
#elif TX_BUFFER_SIZE > 0
  #define SERIAL_FLUSHTX()      SERIAL_OUT(flushTX)
#else
  #define SERIAL_FLUSHTX()
#endif
```
Looks like there is no flushTX for the STM32 HAL?

xC0000005, 2019-12-05:
ESP32 defines a class just to make a flushableSerial...that does not flush. it's literally there to make compile work, which sounds to me like a symptom of something else being broken, but we could tack the same thing in for STM32 if it makes this work.

randellhodges, 2019-12-06:
Not sure if it worth the effort to make the compile work. Then we'll all forget about it and never implement it.

There are some other checks on the STM32 platform for unsupported features, like emergency parser and what-not. Maybe just putting something there might be a better option for now.

reloxx13, 2019-12-06:
https://github.com/MarlinFirmware/Marlin/issues/15235

boelle, 2019-12-11:
just a friendly poke here, but an idea on how we could fix this one?

randellhodges, 2019-12-11:
We could do what @xC0000005 suggested and create a dummy class. That'll prevent the error if someone enables that feature. It "hides" the underlying issue of not having a flushableSerial.

Then we could create a compile warning or something to people it isn't ready yet. The gets rid of the "oh, it won't compile" with a "oh, it isn't ready yet" scenario. (This is optional, but I'd hate to have people turn something on, think it works, when it doesn't really)

If we want to close out the bug, we could implement the workaround as suggested, then open a feature request to implement a flushableSerial for the STM32.

I am not certain what Marlin's philosophy is in this situation. Leaving it as compile bug so we know it needs to be implemented, or hide it until it is implemented. Based on @xC0000005  says about the ESP32 class, I'd say the later.

boelle, 2019-12-11:
i'm not sure either

but yes hiding it seems logical, how would we do that? i'm not good at coding so if i should make a PR then i need to know what to add or replace and where

boelle, 2019-12-11:
of course we need to make sure the core issue is not forgotten

randellhodges, 2019-12-11:
I can use the HAL_ESP32 as a guide and provide a PR for this.

boelle, 2019-12-11:
sounds like a plan then :-)

boelle, 2019-12-24:
was pr #16197 the one that fixes this one?

randellhodges, 2019-12-25:
All good

closed 2019-12-25 by randellhodges