MarlinFirmware/Marlin #16105: [BUG] REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER - Compilation Error

Garyr14, 2019-12-04:
-->

### Bug Description

When I compile BugFix 2.0 using anything but  REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
Enabled, no problem.  However, when I try to compile with that, I get:

avr-gcc: error: CreateProcess: No such file or directory

### My Configurations

Windows 10, Arduino 1.8.10

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3923269/Configuration.zip)

### Steps to reproduce:

Uncomment line:
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
Comment Line:
#define ZONESTAR_LCD

**Expected behavior:** [What you expect to happen]
Successful compile

**Actual behavior:** [What actually happens]
Unsuccessful compile

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

shitcreek, 2019-12-04:
Please use the search function to check if the issue hasn't already been brought up before submitting an issue.
I've found these:
https://github.com/MarlinFirmware/Marlin/issues/7931
https://github.com/MarlinFirmware/Marlin/issues/7990
https://github.com/MarlinFirmware/Marlin/issues/16092

Recommended workaround (thanks to vMeph)

1. Create a new temp folder with a short path, e.g "c:\tmp"

2.  Open Arduino IDE and go to Preferences and see where the preferences.txt file is saved

[image]

in this case it is located in C:\Users\vMeph\AppData\Local\Arduino15

3. Go to that location and open preferences.txt file with Notepad
   On the 3rd line, add this line with the temp folder path created in step 1:

```build.path=C:\tmp```

Garyr14, 2019-12-04:
Thank you.  I should have mentioned that I did in fact search this error and found the cases you referenced.  Unfortunately, none of the suggested solutions have worked so far.

shitcreek, 2019-12-04:
hmm. Have you tried doing a clean reinstallation of Arduino? Best to use VSCode with PlatformIO - it's way better than the Arduino IDE.

Garyr14, 2019-12-04:
I did try a clean reinstallation of Arduino.  I will try VSCode with PlatformIO.  Thank you again.

boelle, 2019-12-04:
or at least move to arduino ide v. 1.9 beta

ellensp, 2019-12-04:
Tried supplied configuration files. Confirmed that REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER is enabled. It compiled fine both in platformio under vscode and in arduino ide 1.8.10 (on linux)

boelle, 2019-12-04:
lets hope @Garyr14 returns with a success

Garyr14, 2019-12-04:
It did compile in platformio under Vscode, but not in arduino ide 1.8.10 (on windows).

boelle, 2019-12-04:
and on 1.9 beta ?

Garyr14, 2019-12-04:
Did not try yet, but can shortly.
For what it's worth, platformio reports problem - unused variable - const bool isHeat = BED_ALT(); in status_screen_DOGM.cpp

ellensp, 2019-12-04:
Are you using the windows app version of arduino ide? As this has been known to cause issues compared with a zip or the exe release.

Garyr14, 2019-12-04:
No - i'm using a zip download version.

ellensp, 2019-12-04:
how about posting the error log?

Garyr14, 2019-12-04:
Sure.  Thanks for the help.

[ErrorLog.txt](https://github.com/MarlinFirmware/Marlin/files/3924210/ErrorLog.txt)

shitcreek, 2019-12-04:
`unused variable` warnings can be ignored.

ellensp, 2019-12-04:
I can confirm this issue on windows. but does not happen on same version of arduino on linux.

ellensp, 2019-12-04:
This seems to be the windows command line length limits for creating a process. It is limited to 32k and the final link command is 36036 characters. Also the old trick of defining the build directory as something short no longer works.

boelle, 2019-12-05:
will close this one as its not errors but warning that can be ignored

closed 2019-12-05 by boelle

ellensp, 2019-12-05:
No. the error is "avr-gcc: error: CreateProcess: No such file or directory" and compilation fails. But i'm not sure there is anything we can do about it. Its a windows limitation.

boelle, 2019-12-05:
if its a windows limit we cant do anything

well we could moan at the arduino ide people since the ball is at their side, but they cant prob not do anything either

emaayan, 2019-12-17:
doesn't this basically mean the marlin 2.0 no longer supports arudino IDE ?

ellensp, 2019-12-19:
This is causing issues for windows users.  If you enable a glcd + TMCsteppers the link command  just get to long for windows. One thing I did notice is that all HAL's get added to the link command when you are only using the one HAL. If you delete all the unused HAL directories from Marlin\src\HAL  This reduces the command enough to allow Arduino on windows to compile.   See eg https://reprap.org/forum/read.php?415,864336,864356#msg-864356