MarlinFirmware/Marlin #16107: [BUG] Update README.md New feature link to dev-2.1.x

arcticlinux, 2019-12-04:
### Bug Description

Update of README.md for release

New Features should referenced the dev-2.1.x branch as the dev-2.0.x branch no longer exists and breaks.

Unsure about other links like the references to the bugfix branch, I don't know enough about how that is handled after release.

```
diff --git a/README.md b/README.md
index aa5ddd6d0..54a22b77d 100644
--- a/README.md
+++ b/README.md
@@ -42,7 +42,7 @@ To build Marlin 2.0 you'll need [Arduino IDE 1.8.8 or newer](https://www.arduino
 ## Submitting Changes
 
 - Submit **Bug Fixes** as Pull Requests to the ([bugfix-2.0.x](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x)) branch.
-- Submit **New Features** to the ([dev-2.0.x](https://github.com/MarlinFirmware/Marlin/tree/dev-2.0.x)) branch.
+- Submit **New Features** to the ([dev-2.1.x](https://github.com/MarlinFirmware/Marlin/tree/dev-2.1.x)) branch.
 - Follow the [Coding Standards](http://marlinfw.org/docs/development/coding_standards.html) to gain points with the maintainers.
 - Please submit your questions and concerns to the [Issue Queue](https://github.com/MarlinFirmware/Marlin/issues).
```

boelle, 2019-12-04:
dev-2.1.x branch is not active yet as far i know

thisiskeithb, 2019-12-04:
There's some discussion (mostly me asking about the two readmes not matching) in PR https://github.com/MarlinFirmware/Marlin/pull/16093, but it was closed. Hopefully it can be looked at again and readmes changed/synced.

arcticlinux, 2019-12-04:
All I know is I happened to click on the link in the readme and it links to https://github.com/MarlinFirmware/Marlin/tree/dev-2.0.x which no longer exists.

boelle, 2019-12-04:
that is because 2.0.x is now the default hence you dont need /tree/dev-2.0.x

thisiskeithb, 2019-12-04:
> that is because 2.0.x is now the default hence you don't need /tree/dev-2.0.x

We're waiting on @thinkyhead for clarification since the readmes don't sync up and obviously one needs to change.

arcticlinux, 2019-12-05:
I'll just close this, sounds like it's in progress.  Just pointing out the dead link, due to the removal of the branch, and offering an alternative, but I'm sure it'll get sorted eventually.

closed 2019-12-05 by arcticlinux