MarlinFirmware/Marlin #16111: Latest Hourly build

hawkeyez88, 2019-12-05:
Try to compile I get this error. Just downloaded the latest build. Running Arduino 1.8.10, tried beta of Arduino and got the same thing. Hopefully it's something I missed.

c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld.exe: C:\Users\John\AppData\Local\Temp\arduino_build_555524/Marlin.ino.elf section `.text' will not fit in region `text'

ellensp, 2019-12-05:
It's a little vague.

boelle, 2019-12-05:
@hawkeyez88 and your error log is way to long, please use a site like paste bin

boelle, 2019-12-05:
can you compile with other versions of arduino IDE?

have you tried vscode+platoform.io ?

hawkeyez88, 2019-12-05:
Thanks @boelle. I'll get the hang of github soon enough. I tried 1.8.8, 1.8.4, and 1.8.10. I have not tried cscode+platform.io. I will do that and report back. Thank You!

boelle, 2019-12-05:
have you tried arduino ide 1.9 beta?

hawkeyez88, 2019-12-05:
That was the first one I tried with.

hawkeyez88, 2019-12-06:
The anet v1 board doesn't show up in vscode+platform.io when I try to import the arduino project, assuming that's what you wanted me to do?

ellensp, 2019-12-06:
You don't import it, you open the folder that has the platformio.ini file.

hawkeyez88, 2019-12-07:
With the latest error on the latest beat version of arduino again I'm back to this error :  
c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld.exe: region `text' overflowed by 7396 bytes

I was reading other pages on this for the past couple of days and haven't gotten anywhere, is it possible the board doesn't have enough free space for the new Marlin :(

ellensp, 2019-12-07:
Times are moving on, 644p based controllers are long dead, 1284p  based controllers (such as the ANET V1.0) are in serous strife.  If you enable advanced features, such as ABL, and GLCDs, It just wont fit any more.  Now It could possibly be made to work, If someone want to put the many hours of of work into optimizing the firmware. Or you could buy a modern controller for $25  dollars.

boelle, 2019-12-07:
as @ellensp said with 644p options are slim to put it mild

https://github.com/boelle/Marlin/tree/OMC-with-SD-and-BEDLEVEL

is a setup for a 644p based printer, i had to disable a lot to make it fit, but i managed to have sd print and manual bed level

but things like saftety etc are gone-ski, i mainly have that setup for my first printer i got, not using it much but still keeping it just in case and for job where quality does not matter

boelle, 2019-12-07:
will close this one as its not error in marlin as such

its a combination of limiting hardware and to many features enabled

closed 2019-12-07 by boelle