MarlinFirmware/Marlin #16114: [FR] Thai System Language

joshuasiumao, 2019-12-05:
###Description
I have tried to translate the system language in Thai so I could help to add Thai language into Marlin.
[language_th.txt](https://github.com/MarlinFirmware/Marlin/files/3925281/language_th.txt)
###Benefits
System Language

###Related Issues

Adding more language to Marlin

ellensp, 2019-12-05:
Ill have a try at a this...   for marlin 2.0    but you will have to check my work as sadly I cant read Thai 
NB the pull request you made is not anything like correct for marlin 2. so basically starting afresh and cutting and pasting your translations.  then will have to work out how to get it on the lcd correctly...  This could take a while...

ellensp, 2019-12-05:
First the limitations
There is no way you can do this on any of the char based displays. So 20x4 are out of the question. They only have 8 re definable characters.You you have to use a 128x64 graphics LCD.

Going to need to make a font for this language...

joshuasiumao, 2019-12-05:
I really appreciate your help! I will try my best to help you. Just tell me what do you need in order to make it up and running.... :)

ellensp, 2019-12-06:
I think this might be a fools errand...  The Thai characters are so squiggly. For eg ญ  (Yo Ying)  Trying to put this into a 12 pixels high by 6 pixels wide character and be readable is just not possible you just get 3 distorted lines.  Several other characters are only different by 1 pixel.     Looking into the character dimensions, is this size actually a limitation or just an adopted standard...

AnHardt, 2019-12-06:
Like the Chinese glyphs. They are at least 12x12 if i remember correctly. Have a look into https://github.com/MarlinFirmware/Marlin/blob/2.0.x/buildroot/share/fonts/marlin-6x12-3.bdf with [FontForge](https://fontforge.org/en-US/) (or similar).

ellensp, 2019-12-07:
Found a nice font that marlin seems to like (after correcting the STARTCHAR values) noting it here for future reference. https://raw.githubusercontent.com/tlwg/thaixfonts/master/etlthai/etl16-thai.bdf

Next problem...   some characters are combined.  eg the Thai string for ready (พร้อม) breaks down to 5 characters in unicode but characters 2 and 3 are meant to be combined see https://qaz.wtf/u/show.cgi?show=%E0%B8%9E%E0%B8%A3%E0%B9%89%E0%B8%AD%E0%B8%A1&type=string  But Marlin of course doesn't know this and prints 5 separate characters....   
See https://pasteboard.co/IK7BOcf.jpg  and https://pasteboard.co/IK7CCxr.jpg

AnHardt, 2019-12-07:
Keywords for that are 'ligatures' and 'kerning'. That can't be managed by u8g-lib itselves. Maybe in u8g2. I remember a discussion [here](https://github.com/olikraus/u8g2/issues/466). Maybe you can backport something to u8g. Oli is not supporting u8g any more (only u8g2) - so feel free to improve our [HAL-version](https://github.com/MarlinFirmware/U8glib-HAL).

[See also](https://github.com/olikraus/u8glib/issues/436)

ellensp, 2019-12-07:
I can set the BBX values to -ve numbers in the font then it prints in the correct place, but then something breaks and it truncates the string being printed that point...   will look into this and the info above.

boelle, 2019-12-07:
think this one is solved when this PR is merged

https://github.com/MarlinFirmware/Marlin/pull/16112

ellensp, 2019-12-07:
@boelle do you even look at it before commenting?, go take a look at the pull request you listed.  90% of this is aliexpress links!  and a bit of eeprom stuff. no language updates, no new fonts. nothing relevant to this other than the misleading title.

boelle, 2019-12-07:
>@boelle do you even look at it before commenting?, go take a look at the pull request you listed. 90% of this is aliexpress links! and a bit of eeprom stuff. no language updates, no new fonts. nothing relevant to this other than the misleading title.

nope but if this is the case i think we should close the PR as spam?

ellensp, 2019-12-07:
That PR is not correct, not even the correct format for marlin 2.   Was the starting point...

boelle, 2019-12-07:
i have put a do not merge label on it and then made a comment

boelle, 2019-12-07:
@ellensp what does it take to add language, a new language file but what more?
i know a font might be needed

trying to do the PR more clean here: https://github.com/MarlinFirmware/Marlin/pull/16136

boelle, 2019-12-07:
@joshuasiumao what language are you best able to translate from?

i have constructed the correct format of language file but you are missing half of what needs to be translated

boelle, 2019-12-07:
@joshuasiumao here is the file but there are a lot of lines that needs translation

all lines that begin with # need translation from english

example:

[image]

but not lines like these:

[image]

line 456,462,468,545,557,569 you must not change

also skip line 586-592

[language_th.h.txt](https://github.com/MarlinFirmware/Marlin/files/3935548/language_th.h.txt)

ellensp, 2019-12-08:
@boelle 
It needs the translation file, a font file and updates to the display code to handle 'ligatures' and 'kerning' (joining of characters) 
I have already done the first two parts...  but haven't created a PR as the last past is still pending (only found it out 8 hours ago! )

boelle, 2019-12-08:
oki, would it make sense to add th font and language file to my PR ? or will you create one?

i have no problem using my PR, just tell me where to place what file

boelle, 2019-12-08:
if you want to create your own PR will it make sense to close mine and the one from @joshuasiumao ??

boelle, 2019-12-08:
@ellensp btw... are your translation a full translate? i found that the one @joshuasiumao did had a lot missing

ellensp, 2019-12-08:
I will create my own when it is ready...  Have Patience 
and ig your really keen look at https://github.com/ellensp/Marlin

boelle, 2019-12-08:
hehe, patience is the keyword in many places

but i will at least close my pr, you think i should do the same with @joshuasiumao's PR?

ellensp, 2019-12-08:
It has a lot missing but is at least in correct format.  https://github.com/ellensp/Marlin/blob/bugfix-2.0.x/Marlin/src/lcd/language/language_th.h   
@joshuasiumao  please download this language file and add all the ones that are missing and check the rest for errors (some things like insert sd where updated to insert media for eg)

ellensp, 2019-12-08:
@boelle Yes please @joshuasiumao's PR

boelle, 2019-12-08:
@ellensp could he use my txt file which should have everything, but lines starting with # needs translation

might make it more easy on @joshuasiumao  what to translate or not

boelle, 2019-12-08:
ahh, i see in your file you just use // where translation is missing?

@sw on discord also offered to help with translation

ellensp, 2019-12-08:
Yes, it changes the colour to a comment colour in most code editors, makes them really obvious.

joshuasiumao, 2019-12-08:
Got it. Will try to do the translation again but since I'm not very good at GitHub so I do not really know how to do the right PR.

ellensp, 2019-12-08:
@joshuasiumao just pate the updated language_th.h  on pastebin and provide a link, ill do the rest

boelle, 2019-12-08:
or even paste it here as a txt?

joshuasiumao, 2019-12-18:
> It has a lot missing but is at least in correct format. https://github.com/ellensp/Marlin/blob/bugfix-2.0.x/Marlin/src/lcd/language/language_th.h
> @joshuasiumao please download this language file and [...]

I wasn't able to download your file. Could please re-upload it again?

closed 2020-04-29 by thinkyhead