MarlinFirmware/Marlin #16116: [BUG] SanityCheck failing existing Delta compile

Stady234, 2019-12-05:
### Bug Description

When trying to compile with the newest release (as of 12-4-2019 @1645 CST) i was receiving the following error::
`#error "Remove DELTA_CALIBRATION_RADIUS and use MIN_PROBE_EDGE to inset the probe area instead."`

### My Configurations
Attached should be my config files as requested:
[TekkieDad-KLP.zip](https://github.com/MarlinFirmware/Marlin/files/3925442/TekkieDad-KLP.zip)

### Steps to Reproduce
Used the config file from 11-7-2019 that is in use on machine..  transferred setting to updated template file... compile as normal.

**Expected behavior:** [What you expect to happen] For compile to be successful as it is with the 11/7 version.

**Actual behavior:** [What actually happens] error message as stated above.

#### Additional Information

Comparing the 11/7 and the 12/4 M2.0BF downloads I found the only "new" lines that are causing issues... ```
#elif ENABLED(DELTA) && defined(DELTA_CALIBRATION_RADIUS)
  #error "Remove DELTA_CALIBRATION_RADIUS and use MIN_PROBE_EDGE to inset the probe area instead."
```
Found on lines 250/251 *when using notepad++..
--
Included is the "copy error" from ArduinoIDE
[Compile Error.txt](https://github.com/MarlinFirmware/Marlin/files/3925469/Compile.Error.txt)

Here should be image of the 12/4 (left) and the 11/7(right) ArduinoIDE attempted compiles.  Both are using the same config files (attached in zip above)
![image](https://user-images.githubusercontent.com/32916064/70211216-2d8be780-16fa-11ea-8513-b19100c4bb12.png)

sjasonsmith, 2019-12-05:
Messages like this are added to Marlin when configuration options change to help guide people how to update their configurations.

Delta settings changed. You no longer specify DELTA_CALIBRATION_RADIUS, it will be auto-calculated using a combination of your MIN_PROBE_EDGE and your nozzle to probe offset. The upside of that is that it will now respect nozzle to probe offsets which are dynamically configured through the M851 command.

Stady234, 2019-12-05:
Can you refer to article/blog/note about this change?  As i don't understand why/how a configuration line is REMOVED from 11/7 to 12/4 and there is no documentation? or i completely missed it...

M851 - XYZ Probe Offset -  http://marlinfw.org/docs/gcode/M851.html  Does nothing to this issue...

If you meant G33 - Delta Auto Calibration, then I would kinda understand as , yes, it will auto calibrate... however, that still does not explain why the change over the last month to now not use a "pre-load" info of dimensions that has been in many previous versions of releases and bugfixes.

sjasonsmith, 2019-12-05:
https://github.com/MarlinFirmware/Marlin/pull/15887

> M851 - XYZ Probe Offset...  Does nothing to this issue... Never had issues with probe offsets...

You may not, but if you tried to configure a probe offset using M851 then use delta calibration/probing routines you would have realized there was a need for change. The most common result would be physical crashing of the effector into various parts of the printer.

In fact, even G33 did not work as expected with an offset probe, and could easily be made to crash into towers, even with a statically defined probe offset.

In the process of resolving the issues mentioned in this pull request it became clear that this measurement was redundant. We already have a printable radius. We already have a MIN_PROBE_EDGE. We already have a nozzle to probe offset. The calibration radius could simply be computed from these, which added the benefit of supporting dynamic probe changes using M851.

> As i don't understand why/how a configuration line is REMOVED from 11/7 to 12/4 and there is no documentation?

Many options changed in Marlin 2.0, and are not all documented yet. The error messages in SanityCheck.h are intended to provide some guidance to their resolution. In this case the instructions seem fairly clear, and you could also look at any of the delta examples which were updated according tot his change.

I am sure your contributions to the Marlin Documentation project would be more than welcome, which may help future users who encounter this and other issues.

Stady234, 2019-12-05:
> You may not, but if you tried to configure a probe offset using M851 then use delta calibration/probing routines you would have realized there was a need for change. The most common result would be physical crashing of the effector into various parts of the printer.

I guess that may be why i never had that issue be identified.  I have never setup printer in that order.. I have always used G33, G33 P1 (redundant i know), M851 (calibration), Bed level routine;  It is good to hear that the G33 will work better, but i have not had any issues with it so far....

> The error messages in SanityCheck.h are intended to provide some guidance to their resolution.

The only problem I have with that is in this particular case, it did not make sense why to just "remove" when all other examples in the Nov Build group all had that same line, and then all of the examples in Dec Build group did not.  If it was referenced a bit better on that value was now deprecated would have made more sense.  The fact that I am defining an option that will be updated via calibration should not have halted the build.  For example, what if i defined all of those values (from a known good build/calibration) then uploaded this firmware, and never used G33 ?  I can't due to the new SanityCheck entry.

Either way... thank you for clarifying the reasons for the error message.  I have updated the new config file to omit that line, and everything passed... now to upload the firmware and move one!!  lol

Thanks for the help!  

p.s. If i came across rude in previous reply... it was at 1am... and i do apologize.

sjasonsmith, 2019-12-05:
> For example, what if i defined all of those values (from a known good build/calibration) then uploaded this firmware, and never used G33 ? I can't due to the new SanityCheck entry.

You can still specify your geometry manually. The calibration radius was simply the radius at which probing was performed for auto-calibration, not a part of the geometry impacting kinematic calculations.

If you feel the message could be re-worded to make it less confusing for other users, please feel free to post a pull request to change it.

If you are now satisfied with the state of things, could you go ahead and close this issue?

closed 2019-12-05 by Stady234