MarlinFirmware/Marlin #16122: [BUG] ABL probe not homing in the centre of the bed

LordBBQ, 2019-12-05:
Whenever I run an Autohome, the BLTouch  tries to probe off the bed, which I need to stop to avoid imminent disaster. I have enabled bed safe homing. It still does not work properly. 
[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/3928904/Archive.zip)

boelle, 2019-12-05:
what version of marlin?

Vertabreak, 2019-12-05:
Sounds like bed offsets / max travel / probe offsets is incorrect. The user should verify and calibrate all 3 to ensure proper function

thisiskeithb, 2019-12-05:
> what version of marlin?

2.0 according to the configs.

@LordBBQ: You need to enable one of the `AUTO_BED_LEVELING` options since they're all commented out:

```cpp
//#define AUTO_BED_LEVELING_3POINT
//#define AUTO_BED_LEVELING_LINEAR
//#define AUTO_BED_LEVELING_BILINEAR
//#define AUTO_BED_LEVELING_U
```

`AUTO_BED_LEVELING_BILINEAR` would be a good one to start with.

LordBBQ, 2019-12-06:
Is that right for a 3x3 bltouch grid?

thisiskeithb, 2019-12-06:
> Is that right for a 3x3 bltouch grid?

You can do a 3x3 grid with `AUTO_BED_LEVELING_BILINEAR`. In fact, I run it with 7x7.

LordBBQ, 2019-12-06:
Still no joy. Set it to a BiLinear 4x4 grid, and it still does not move.

LordBBQ, 2019-12-06:
Not sure if this is helpful, but when I auto homed on my old Melzi board running TH3D Marlin 1.1.9, when it homed, it hit the switch, came back ~10mm, and hit the switch on all X, and Y. Also, the Z does not move up when deploying the BLTouch. It now just moves until it triggers once.

LordBBQ, 2019-12-07:
> Sounds like bed offsets / max travel / probe offsets is incorrect. The user should verify and calibrate all 3 to ensure proper function. Also the defined steps xyze all look strange for the model. On further examination the steps could be the biggest issue unless it's been modified to be that way physically. Missing a bed leveling type doesn't help either but would have no effect on homing. Steps { 320 , 320, 1600, 372, 3348 } seems very strange to me.

X, Y were originally 80, but it is set to 1/64 microstepping. Z was originally 400, E0 is a normal extruder, E1 is a geared ~3/1.

boelle, 2019-12-07:
@LordBBQ since 2.0 was just released a few days ago has this changed this issue at all?

LordBBQ, 2019-12-07:
I will recompile today off this release and see if the issue is fixed.

LordBBQ, 2019-12-09:
Thanks so much! It is homing fine now. Not sure if this is at all related to the firmware upgrade, but when the drivers enter spread cycle mode, they make a lot of noise, and a loud hum.

Vertabreak, 2019-12-10:
that is the trade off using spreadcycle louder but more torque.

boelle, 2019-12-10:
will close this one as its fixed, tmc are not as silent in spreadcycle as in stealthchop

as @Vertabreak said there is a tradeoff

closed 2019-12-10 by boelle