MarlinFirmware/Marlin #16127: About Printer Menu Missing

venuasg, 2019-12-06:
In Marlin 2.0.x "About Printer" menu is missing on the lcd, is it i need to enable something to get it?
Also i would like to add a custom menu, can anyone guide how to go about it.

Thanks in advance.

ellensp, 2019-12-06:
Not really the place or this... this is for firmware bugs and feature requests.

About printer needs #define PRINTCOUNTER in Configuration.h 
And see https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/Configuration_adv.h#L2590  for custom menu items.

venuasg, 2019-12-06:
Thank's, will take care.
However after enabling #define PRINTCOUNTER, i still could not see the About Printer menu option.

ellensp, 2019-12-06:
Opps needs #define LCD_INFO_MENU in Configuration_adv.h,  PRINTCOUNTER is optional under About printer, if you want it.

venuasg, 2019-12-06:
Thanks, Got it and working. Now i can see the menu.

closed 2019-12-06 by venuasg