MarlinFirmware/Marlin #16129: [BUG] core_xy G38 probe

amin7, 2019-12-06:
### Bug Description

for core_xy G38.2 does not work

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3933690/Marlin.zip)

### Steps to Reproduce

1. power up & home
2. send G38.2 Z10
3. close probe trigger

**Expected behavior:** [What you expect to happen]
stop moving
**Actual behavior:** [What actually happens]
continue moving
send error "Error:Failed to reach target"
#### Additional Information
src/module/endstops.cpp

#if ENABLED(G38_PROBE_TARGET) && PIN_EXISTS(Z_MIN_PROBE) &**& !(CORE_IS_XY || CORE_IS_XZ)**
looks like issue is there
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

boelle, 2019-12-11:
@amin7 

src/module/endstops.cpp

what line do you look at?

and do you know what to change it to that solves the problem? if so i could make a PR and send it in

amin7, 2019-12-12:
hi @boelle ,

src/module/endstops.cpp lines 506 & line 689

i have coreXY and to activate probe i removed CORE_IS_XY

>  #if ENABLED(G38_PROBE_TARGET) && PIN_EXISTS(Z_MIN_PROBE) && !(CORE_IS_XZ)

work's fine.
but i do not investigate it deaply, problably there was some reason why it is there.

shitcreek, 2019-12-19:
I am guessing maybe it should have been **CORE_IS_YZ** instead.
@Bob-the-Kuhn - what s your thought on this?

Vertabreak, 2019-12-19:
&& !(CORE_IS_XY || CORE_IS_XZ) = and not (core is xy) or (core is xz)
Not real familiar with corexy but it does seem odd.

boelle, 2019-12-24:
btw, who can confirm there is an issue? use the same configs as OP and the same hardware if possible

amin7, 2019-12-26:
Hi @boelle , yes (this part of code was not changed 2 month)

boelle, 2020-01-08:
@Bob-the-Kuhn any idea on this one?

boelle, 2020-01-23:
we can only hope that @Bob-the-Kuhn have and idea on what is going on

boelle, 2020-01-30:
@amin7 has anything changed?

@Bob-the-Kuhn any idea what is going on here? @shitcreek @thinkyhead

amin7, 2020-02-01:
@boelle 
pull today - behaviour is the same
ps i am using (2.0.x) brunch

boelle, 2020-02-08:
@shitcreek @thinkyhead any idea what is going on here?

thinkyhead, 2020-02-11:
Yes, I can confirm that `G38` has not been implemented for CoreXY or Delta.

thinkyhead, 2020-02-11:
Hmm, on further examination it looks like the attempt was made. The probe pin is being monitored during the `G38` move….

So it works fine on regular cartesian machines @amin7 but not on your core machines?

amin7, 2020-02-11:
@thinkyhead clean copy -on my core_XY does not work.
if remove defines that I mention -work's fine.

amin7, 2020-04-10:
Version 2.0.5.3. config the same, issue present

thinkyhead, 2020-05-01:
It is still not implemented. Someone with hardware like yours will need to work on this.

olemin7, 2020-06-24:
today origin/bugfix-2.0.x has issue

olemin7, 2020-06-24:
fix path
diff --git a/Marlin/src/module/endstops.cpp b/Marlin/src/module/endstops.cpp
index 47d63df64..a08335620 100644
--- a/Marlin/src/module/endstops.cpp
+++ b/Marlin/src/module/endstops.cpp
@@ -498,7 +498,7 @@ void Endstops::update() {
   #define UPDATE_ENDSTOP_BIT(AXIS, MINMAX) SET_BIT_TO(live_state, _ENDSTOP(AXIS, MINMAX), (READ(_ENDSTOP_PIN(AXIS, MINMAX)) != _ENDSTOP_INVERTING(AXIS, MINMAX)))
   #define COPY_LIVE_STATE(SRC_BIT, DST_BIT) SET_BIT_TO(live_state, DST_BIT, TEST(live_state, SRC_BIT))
 
-  #if ENABLED(G38_PROBE_TARGET) && PIN_EXISTS(Z_MIN_PROBE) && !(CORE_IS_XY || CORE_IS_XZ)
+  #if ENABLED(G38_PROBE_TARGET) && PIN_EXISTS(Z_MIN_PROBE) && !(CORE_IS_XZ)
     // If G38 command is active check Z_MIN_PROBE for ALL movement
     if (G38_move) UPDATE_ENDSTOP_BIT(Z, MIN_PROBE);
   #endif
@@ -736,7 +736,7 @@ void Endstops::update() {
     #define PROCESS_ENDSTOP_Z(MINMAX) PROCESS_DUAL_ENDSTOP(Z, MINMAX)
   #endif
 
-  #if ENABLED(G38_PROBE_TARGET) && PIN_EXISTS(Z_MIN_PROBE) && !(CORE_IS_XY || CORE_IS_XZ)
+  #if ENABLED(G38_PROBE_TARGET) && PIN_EXISTS(Z_MIN_PROBE) && !(CORE_IS_XZ)
     #if ENABLED(G38_PROBE_AWAY)
       #define _G38_OPEN_STATE (G38_move >= 4)
     #else

sjasonsmith, 2020-07-04:
@olemin7 it looks like you pasted a patch but it got rendered as Markdown and messed up. Does that fix the issue? If so, is it in a state that you could open a pull request to contribute the patch?

olemin7, 2020-07-04:
some thing gone wrong with pr
i have attached file,On branch bugfix-2.0.x
[endstops.zip](https://github.com/MarlinFirmware/Marlin/files/4873083/endstops.zip)

closed 2020-08-09 by github-actions[bot]