MarlinFirmware/Marlin #16132: [BUG] Marlin v2.0.x : Endstops do not trigger on RAMPS

T-SquaredProductions, 2019-12-06:
### Bug Description

I tried to autohome my printer, but even with the endstops attached and their LEDs activating when hit, the printer will not properly home.

### My Configurations
[MARLINCONFIG.zip](https://github.com/MarlinFirmware/Marlin/files/3934414/MARLINCONFIG.zip)

### Steps to Reproduce

1. Use RAMPS 1.6
2. Go to 'Motion'>'Auto Home'

**Expected behavior:** I expected the printer to autohome to the minimum coordinates at the front left of the heating bed.

**Actual behavior:** The z-axis rises, the bed moves toward the user, and the z-axis rises again, as if the endstops have already been triggered.

boelle, 2019-12-07:
check your endstops with M119

ie make sure they are all NOT activated, then send M119 and see what the printer thinks

then put your finger on each endstop and send M119 again

boelle, 2019-12-07:
@T-SquaredProductions since 2.0 was just released a few days ago has this changed this issue at all?

boelle, 2019-12-08:
@T-SquaredProductions did you check the endstops?

T-SquaredProductions, 2019-12-10:
> @T-SquaredProductions did you check the endstops?

Pronterface reports all of them are triggered when they are not. Do I need to invert their logic?

ManuelMcLure, 2019-12-10:
And they report open when you do trigger them? If so, then yes - you need to invert the logic.

T-SquaredProductions, 2019-12-10:
Ok, Endstop problems fixed. They report normally. But now I have another problem that led to this: The z-axis motor moves up normally, but after that, motors do not engage or move, and after a few seconds, the printer halts with the usual "ERROR: Printer Halted. Please Reset." message. Pronterface reports a "Error:Printer halted. kill() called!" echo message.

boelle, 2019-12-10:
better you hit close on this one and create a new since problem is not the same and config also have changed

closed 2019-12-10 by T-SquaredProductions